ROBOTIS-GIT/open_manipulator #1: CAD data of open_manipulator.

roboshiba, 2017-03-17:
Hi, Lim-san. I have questions.

(1) Is CAD data of open_manipulator exposed on OnShape?

(2) The shape of the end effector of the manipulator changes according to the application.
Can you Branch data of user created frame or end effector?

routiful, 2017-03-17:
Hi, shibata san. 

Open_manipulator will be exposed OnShape when we test it over. Users can export it in near the future :)

The end-effector can be changed depending on application. We have plan to support other tool tips but ,if they want, they can create their own grippers what they need.

Thanks.

roboshiba, 2017-03-24:
Hi, Lim-san.
Thanks.
I also do design improvement of manipulator. I want to share it.
For example, a lightweight hinge cover :)==>[https://cad.onshape.com/documents/6037ebcd9a4306846b0618c9/w/c9f93c4e1150c41885db4dad/e/de27c63cab5c7ae541ba66ab](url)

closed 2017-03-24 by roboshiba