ROBOTIS-GIT/open_manipulator #6: About Open Manipulator parts and Assembled selling.

roboshiba, 2017-08-18:
Is it okay for a third party to sell Open Manipulator parts and cron?

routiful, 2017-08-18:
Hi :)

OpenManipulator is not selling product yet.
But you can sell Dynamixel and other parts for third party.
If they want to any supports for operating OpenManipulator with ROS or OpenCR,
they could open a ticket in Github issue page.

Thanks.
Darby

roboshiba, 2017-08-21:
Thank you for reply.

I am writing this Issue as open case study for the Open Manipulator to grow all over the world.

There is no notation of License in the Repository, but what kind of licenses are the design data of the Open Manipulator parts and assemblies?
OnShape interprets Public's data as license free.
It's the same?

routiful, 2017-08-21:
Main software of OpenManipulator is released under the Apache 2.0 license
and Hardware is Open Source Hardware Statement of Principles and Definition v1.0.

roboshiba, 2017-08-24:
Thank you very much. 
I will contact you again when concrete cases occur.

closed 2017-08-24 by roboshiba