ROBOTIS-GIT/open_manipulator #9: open_manipulator_dynamixel_controller error message

orasman, 2017-09-12:
Hi~ 
I beginner of ROS.
open_manipulator package is a great reference to start manipulator!
Thanks for ROBOTIS team!

When I follwed dynamixel control example, there is a error as below.

========
PARAMETERS
 * /baud_rate: 57600
 * /device_name: /dev/ttyUSB0
 * /protocol_version: 1.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.7

==========

"Sync Read Failed!" error message generated from 
from "_Dynamixel_controller.cpp_"
_bool DynamixelController::readDynamixelState(void)_ function

_bool DynamixelController::control_loop()
{
  // Read & Publish Dynamixel position
  **//readDynamixelState();    //comment out for** 
}_

routiful, 2017-09-12:
Hi :)

The error message shows out of range of vector size. 
Did you set Dynamixel ID 1, 2, 3 ??

orasman, 2017-09-12:
Yes, I did.
I set my dynamixel ID as 1,2,3.

routiful, 2017-09-12:
Oh, you have used protocol 1.0.
Protocol 1.0 doesn't support **sync_read**.
If you want to read present_position of Dynamixel, displace Dynamixel using protocol 2.0 or change function sync_read to readRegister.

orasman, 2017-09-13:
i got it! thank you!
if you have plan to update your package to support v1.0 protocol, then i will wait.
if not, i have to do it bu myself.
but if i have trouble with that, i will ask to help again!
thanks a lot!

routiful, 2017-09-13:
I will update it as soon as possible.

Thank you for your interest :)

closed 2017-09-13 by routiful

MarcoDotIO, 2020-02-19:
> Oh, you have used protocol 1.0.
> Protocol 1.0 doesn't support **sync_read**.
> If you want to read present_position of Dynamixel, displace Dynamixel using protocol 2.0 or change function sync_read to readRegister.

Hello there routiful,

I would like some clarification with this issue.

In my case, I daisychained two MX-106 motors (ID 2 and 3) for the arms, with another one (ID 1) controlling the head of a project I'm working on. This error only happens with motor IDs 2 and 3 and works completely fine with motor ID 1.

I would like you to clarify on the regard of "displace Dynamixel using protocol 2.0" regarding what change I need for my code.

And currently, the changes I have with the workbench operator are heavy, though I haven't touched any source code for the controller. I have linked the motor ID configuration files, along with the modified workbench joint operator code, in this comment.

[yaml_and_cpp_files.zip](https://github.com/ROBOTIS-GIT/open_manipulator/files/4223188/yaml_and_cpp_files.zip)

Please respond ASAP if there are any questions / concerns / issues here.