ROBOTIS-GIT/open_manipulator #10: Code for protocol1.0 dynamixel servos.?

gokul-gokz, 2017-09-18:
Can you share the updated code for dynamixel controller which supports Dynamixel 1.0 in the open manipulator .

orasman, 2017-09-18:
Hello~ it's me again.
How long does it takes to upgrade code for protocol v1.0?
I have to finish my manipulator project before end of September.
I need your help!
my manipulator already made with multiple EX-106R. It's too expensive change to MX-106R.

every night, I look into source code, but it is not easy to understand framework of dynamixel_workbench_toolbox including multi_driver.cpp.
also i think about not modifying "multi_driver.cpp"(dynamixel_workbench_toolbox) but simply modifying "dynamixel_controller.cpp" (open_manipulator package)
especially, DynamixelController::readDynamixelState(void) function using syncReadPosition, replace to using readRegister.
But it doesn't derived from DynamixelDriver. so it is not simple solution!

If you are available, please consider upgrade package to support V1.0 protocol.
Thank you!

routiful, 2017-09-19:
Hi :)

I am preparing OpenManipulator for protocol v1.0.
I think it will be released in october.

Thanks

orasman, 2017-09-19:
I fixed code for protocol V1.0 (temporary solution)
by modifying "dynamixel_controller.cpp" at "open_manipulator_dynamixel_ctrl" folder.
replace "syncReadPosition" to "readMultiRegister".
I changed only two lines as below.
it looks like simple, but it was hard to understood workbench framework of Dynamixel SDK for me.

 bool DynamixelController::readDynamixelState(void) 
 {
      readValue_->pos.clear();
     //******* change here **********  //this is works for V1.0
     if (!multi_driver_->readMultiRegister("present_position")) 
          ROS_ERROR("readMultiRegister Read Failed!");

    //below 2 lines was original code : not support V1.0 protocol
    //if (!multi_driver_->syncReadPosition(r  eadValue_->pos))  //original code : not support V1.0 protocol
    //  ROS_ERROR("Sync Read Failed!");

    sensor_msgs::JointState dynamixel_position;

    for (int id = 1; id <= MAX_DXL_NUM; id++)
    {
        std::stringstream id_num;
        id_num << "id_" << id;
        dynamixel_position.name.push_back(id_num.str());

      //******* change here **********  //this is works for V1.0
      dynamixel_position.position.push_back(convertValue2Radian((int32_t)   multi_driver_->read_value_["present_position"]->at(id-1)     ));
    
      //below line was original code : not support V1.0 protocol
      //dynamixel_position.position.push_back(convertValue2Radian((int32_t)readValue_->pos.at(id-1)));
    }

gokul-gokz, 2017-09-21:
Thanks for your help. Now I am able to read the present states of the actuators.

gokul-gokz, 2017-09-21:
@orasman Are you able to control the real hardware through moveit by modifying this code?

orasman, 2017-09-22:
@gokul-gokz Yes i can. because syncWritePosition() method work for protocol V1.0 also. 
So if you launch position_ctrl and launch moveit, then click plan&execute, then motor rotating.

I suggest you should try dynamixel_workbench package working for your dynamixel servo.
it has single_manager to check your dynamixel status. 
position_control to control two dynamixel. 
if it work correctly, then you can goto next step : open_manipulator.

gokul-gokz, 2017-09-22:
@orasman
Yes. I have modified the open manipulator codet to work with my 6DOF manipulator. But the trajectory executed is not smooth in the real hardware and it is very jerkier. Since the position_control node in open manipulator reads the trajectory positions from move_group->display_planned_path variable and executes it. How to smoothen it? @routiful

orasman, 2017-09-22:
@gokul-gokz i have no idea about how to make the motion smooth. 
i'm just start with open_manipulator now. 
maybe you are far more then me.
i also experienced similar jurky movement but not so wired in my case. 
i need to solve another issues, so i should take care jurky issue later.

orasman, 2017-09-24:
@routiful i encounter same problem like gokul-gokz. motion is very jurky. i tried change options in moveit pannel, but it not help. so you have any idea gow to make motion smoother?

gokul-gokz, 2017-10-05:
How to smoothen the trajectory?@routiful

routiful, 2017-10-05:
Hi @gokul-gokz and @orasman :)

This week is holiday in my country. I will check it your issue next week and comment ASAP.

Thanks

routiful, 2017-10-11:
Hi :) I'm back.

@orasman found proper function and modified code for present position of Dynamixel. It was nice!

To make motion smoother, I used an **industrial_trajectory_filters**. It is already applied OpenManipulator moveit package. 

Follow the instruction below.
1. Download ROS_INDUSTRIAL_CORE package on your workspace
`$ git clone https://github.com/ros-industrial/industrial_core.git`

2. Check industrial_trajectory_filters
` $ cd ~/{$my_ws}/src/industrial_core/industrial_trajectory_filters/`

3. Copy src, include folder and xml file to moveit package
```
$ cp -r planning_request_adapters_plugin_description.xml src include ~/catkin_ws/src/open_
manipulator/open_manipulator_moveit_example/
```

4. Create new yaml file in config folder and copy below coefficient
```
$ cd ~/{$my_ws}//src/open_manipulator/open_manipulator_moveit/config
$ gedit smoothing_filter_params.yaml
```
```
smoothing_filter_name: /move_group/smoothing_5_coef
smoothing_5_coef:
- 0.25
- 0.50
- 1.00
- 0.50
- 0.25
```

5. Open ompl_planning_pipeline.launch.xml in launch folder and add below filter
```
$ cd ~/catkin_ws/src/open_manipulator/open_manipulator_moveit_example/launch
$ gedit ompl_planning_pipeline.launch.xml
```
```
industrial_trajectory_filters/UniformSampleFilter
industrial_trajectory_filters/AddSmoothingFilter
```

6. Add parameter in the same xml file
```
<param name="sample_duration" value="0.04" />
<rosparam command="load" file="$(find open_manipulator_moveit)/config/smoothing_filter_
params.yaml"/>
```
Below xml file is example of set OpenManipulator

```
<launch>

  <!-- OMPL Plugin for MoveIt! -->
  <arg name="planning_plugin" value="ompl_interface/OMPLPlanner" />

  <!-- The request adapters (plugins) used when planning with OMPL.
       ORDER MATTERS -->
  <arg name="planning_adapters" value="
               industrial_trajectory_filters/UniformSampleFilter
               industrial_trajectory_filters/AddSmoothingFilter
               default_planner_request_adapters/AddTimeParameterization
               default_planner_request_adapters/FixWorkspaceBounds
               default_planner_request_adapters/FixStartStateBounds
               default_planner_request_adapters/FixStartStateCollision
               default_planner_request_adapters/FixStartStatePathConstraints" />

  <arg name="start_state_max_bounds_error" value="0.1" />

  <param name="planning_plugin" value="$(arg planning_plugin)" />
  <param name="request_adapters" value="$(arg planning_adapters)" />
  <param name="start_state_max_bounds_error" value="$(arg start_state_max_bounds_error)" />

  <param name="sample_duration" value="0.04" />

  <rosparam command="load" file="$(find open_manipulator_moveit)/config/ompl_planning.yaml"/>
  <rosparam command="load" file="$(find open_manipulator_moveit)/config/smoothing_filter_params.yaml"/>

</launch>
```

After that, you can show smoother motion when your manipulator is controlled.
If you have any problem, please feel free to reply on this issue page.

Thanks all
Darby

routiful, 2018-04-24:
This issue will be closed since there were no actions for a while. 
For more information about smoothing path for joint control, please refer to Chapter13_Manipulator(page 443~444) in [ROS Robot Programming](https://community.robotsource.org/t/download-the-ros-robot-programming-book-for-free/51/38).

closed 2018-04-24 by routiful