ROBOTIS-GIT/open_manipulator #11: Controlling real hardware

gokul-gokz, 2017-09-21:


ROBOTIS-Will, 2017-09-25:
Hello gokul-gokz,
Please read [this thread](https://answers.ros.org/question/250855/no-known-controllers-and-their-joints-in-moveit-when-connecting-real-robot/) and see if it helps to solve your problem.

routiful, 2017-10-10:
Hi @gokul-gokz , sorry for late reply

SRDF is configured by setup assistant based on URDF.
After changing an URDF for your robot, you have to modified open_manipulator_position_ctrl package and open_manipulator_dynamixel_ctrl package.
These packages include move_node and dynamixel_workbench. You need to check define the number of dynamixels and joints.
This number is related with error message of Eigen lib.

Thanks
Darby

closed 2017-12-11 by routiful