ROBOTIS-GIT/open_manipulator #19: Turtlebot3+OpenManipulator : XM430-W350-T or XM430-W350-R ?

roboshiba, 2018-02-13:
I am making an OpenManipulator in Japan.
Questions when installing OpenManipulator on Turtlebot3.

Actuator of Turtlebot3 is XL430-W250-T and XM430-W210-T. This is a TTL model.
OpenManipulator is XM430-W350-R. This is the RS485 model.
Even when installing OpenManipulator on TB3, is RS485 model official recommended?
Are there any issue when using the TTL model?

[image]

routiful, 2018-02-13:
Hello Shibata-san :)

I have tested Dynamixel X to use TTL communication but it has not any trouble. 
However, due to reliability, we officially recommends to use RS-485 version of Dynamixels.
Because OpenCR supports TTL and RS-485, TB3 and OpenManipulator could be worked in same controller.

Best regards
Darby

roboshiba, 2018-02-13:
Thank you for quick reply.

I understood your policy.
If software resource,example does not differ between TTL and RS485, it is no problem to user.

The use of XL430(OpenManipulator Lite version) is also assumed.

closed 2018-02-13 by roboshiba