ROBOTIS-GIT/open_manipulator #21: REQUIRED process [open_manipulator_dynamixel_controller-2] has died!

MorrisJeong, 2018-03-07:
When I launch that code, the following error occur.

   roslaunch open_manipulator_dynamixel_ctrl dynamixel_controller.launch

I already let the USB accss allow, but it is not working too.
What would I do to solve this error??

started roslaunch server http://localhost:*****/

SUMMARY
========

PARAMETERS
 * /baud_rate: 1000000
 * /device_name: /dev/ttyUSB0
 * /rosdistro: kinetic
 * /rosversion: 1.12.12
 * /scan_range: 10

NODES
  /
    open_manipulator_dynamixel_controller (open_manipulator_dynamixel_ctrl/dynamixel_controller)

auto-starting new master
process[master]: started with pid [5590]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to *****
process[rosout-1]: started with pid [5603]
started core service [/rosout]
process[open_manipulator_dynamixel_controller-2]: started with pid [5609]
[ INFO] [1520408424.601165791]: open_manipulator_dynamixel_controller : Init OK!
================================================================================REQUIRED process [open_manipulator_dynamixel_controller-2] has died!
process has died [pid 5609, exit code -11, cmd /home/jeong/catkin_ws/devel/lib/open_manipulator_dynamixel_ctrl/dynamixel_controller __name:=open_manipulator_dynamixel_controller __log:=/home/jeong/.ros/log/.../open_manipulator_dynamixel_controller-2.log].
log file: /home/jeong/.ros/log/.../open_manipulator_dynamixel_controller-2*.log
Initiating shutdown!
================================================================================
[open_manipulator_dynamixel_controller-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

MorrisJeong, 2018-03-08:
I used MX-106 motors and the protocol version of those is 2.0. And I updated all the motors.
Is there anything else I can do for solving this problem?

MorrisJeong, 2018-03-11:
I have two motors. And I set the ID to one and two separately.

routiful, 2018-03-11:
Would you mind trying to connect Dynamixel by [Dynamixel-workbench](https://github.com/ROBOTIS-GIT/dynamixel-workbench)??

It is more convenient for debugging

Thanks
Darby

routiful, 2018-04-16:
This issue have been wanting a help from long time ago, but still no action is shown. I leave the question label, but want an understanding of closing this issue. We will reopen the issue if any actions occur.

unverciftci, 2018-05-14:
I have a similar error with XL-430 and OpenCR for open manipulator:

setting /run_id to 66eacd0a-578b-11e8-b675-60f677ef5434
process[rosout-1]: started with pid [4654]
started core service [/rosout]
process[dynamixel_controller-2]: started with pid [4663]
[PortHandlerLinux::SetupPort] Error opening serial port!
[PortHandlerLinux::SetupPort] Error opening serial port!
================================================================================REQUIRED process [dynamixel_controller-2] has died!
process has died [pid 4663, exit code -11, cmd /home/unver/catkin_ws/devel/lib/open_manipulator_dynamixel_ctrl/dynamixel_controller __name:=dynamixel_controller __log:=/home/unver/.ros/log/66eacd0a-578b-11e8-b675-60f677ef5434/dynamixel_controller-2.log].
log file: /home/unver/.ros/log/66eacd0a-578b-11e8-b675-60f677ef5434/dynamixel_controller-2*.log
Initiating shutdown!
================================================================================
[dynamixel_controller-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

routiful, 2018-05-15:
Hi :)

you have to give permission to open port

unverciftci, 2018-05-15:
Hi,
I tried

sudo chmod a+rw /dev/ttyUSB0

but it did not work.

routiful, 2018-05-16:
You can search connected usb port in your laptop.
Please check it and try again.

`$ ls /dev`

closed 2018-05-22 by robotpilot