ROBOTIS-GIT/open_manipulator #30: open_manipulator dynamixel_controller.launch crashing

gokul-gokz, 2018-05-05:
I have 6 dynamixel pro motors connected in daisy chain. When I launch the dynamixel controller node in the open manipulator package, the node throws an error

 
started roslaunch server http://asimov-All-Series:37783/

`SUMMARY
========

PARAMETERS
 * /baud_rate: 57600
 * /device_name: /dev/ttyUSB0
 * /rosdistro: kinetic
 * /rosversion: 1.12.7
 * /scan_range: 253v

NODES
  /
    open_manipulator_dynamixel_controller (open_manipulator_dynamixel_ctrl/dynamixel_controller)

auto-starting new master
process[master]: started with pid [4429]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 5a1045d8-501f-11e8-abb0-e8de270f8bfc
process[rosout-1]: started with pid [4442]
started core service [/rosout]
process[open_manipulator_dynamixel_controller-2]: started with pid [4449]
[ INFO] [1525495640.602282023]: open_manipulator_dynamixel_controller : Init OK!
================================================================================REQUIRED process [open_manipulator_dynamixel_controller-2] has died!
process has died [pid 4449, exit code -11, cmd /home/asimov/open_manipulator_ws/devel/lib/open_manipulator_dynamixel_ctrl/dynamixel_controller __name:=open_manipulator_dynamixel_controller __log:=/home/asimov/.ros/log/5a1045d8-501f-11e8-abb0-e8de270f8bfc/open_manipulator_dynamixel_controller-2.log].
log file: /home/asimov/.ros/log/5a1045d8-501f-11e8-abb0-e8de270f8bfc/open_manipulator_dynamixel_controller-2*.log
Initiating shutdown!
================================================================================
[open_manipulator_dynamixel_controller-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
`

The issue is similar to the below issue
#https://github.com/ROBOTIS-GIT/open_manipulator/issues/28

routiful, 2018-05-09:
Hi :)

Each[ Dynamixel PRO](http://emanual.robotis.com/docs/en/dxl/pro/h54-200-s500-r/) needed 24V nominal voltage. But if its has been connected 3 more, voltages drop can occur. 

Hence, you would try to connect another power cable through fourth PRO.

Thanks

gokul-gokz, 2018-05-09:
The manipulator has 3 of the PRO-H54-100-S500-R(Base-shoulder-elbow). All these 3 actuators has separate power cable of 24.0V.
And then 3 of the PRO-H42-20-S300-R(Roll-pitch-Roll). For these 3 actuators there is a single power cable of 24.0V.
Then there is a gripper with MX-64 protocol 2.0. There is a separate 12.0V power cable.

Things tried so far:
1. All the actuators are powered. All the actuators are detected individually using dynamixel workbench single manager node.
 
2. When I ran the dynamixel controller node of open manipulator, to detect all the actuators, the node crashes with the error stated above.

3. All the actuators are powered. But changed the scan range to detect only 3 of the PRO-H54-100-S500-R(Base-shoulder-elbow) actuators. The actuators got detected and I was able to command them to different goal positions.

4.All the actuators are powered. And the changed the dynamixel id's and scan range to detect 2 of the PRO-H54-100-S500-R and 1 of the PRO-H42-20-S300-R. The actuators got detected and I was able to command them to different goal positions.

5.All the actuators are powered. And changed the scan range to detect 4 actuators( 3 of the PRO-H54-100-S500-R and 1 of the  PRO-H42-20-S300-R).  The node crashed with the error stated above.

6.All the actuators are powered. And changed the scan range to detect 4 actuators( 3 of the PRO-H54-100-S500-R and 1 of the MX-64).  The node crashed with the error stated above.

What might be the possible issue?

routiful, 2018-05-10:
I checked [Manipulator-H ](http://emanual.robotis.com/docs/en/platform/manipulator_h/introduction/) by launching Dynamixel-workbench Controller. It working fine. 

This Dynamixel PRO was set 2000000 Baudrate and connect power cable separatesly.

gokul-gokz, 2018-05-10:
@routiful  I tried setting the baud rate to 2000000. But still the same issue exists. It is only able to detect 3 motors at a time. I have attached the electrical  diagram of the manipulator.  [https://cdn1.imggmi.com/uploads/2018/5/10/fcb2913e62ea506c1253179ef7e7530f-full.png](url) Check it to see any probable issues.

I have also tested it by creating a controller manager in ROBOTIS framework, but it resulted in error that i have reported #https://github.com/ROBOTIS-GIT/ROBOTIS-Framework/issues/62

routiful, 2018-05-14:
Sorry i can't open attached png file. Would you check it?

gokul-gokz, 2018-05-14:
I have updated the link https://ibb.co/bUGQFd . Have a look at this.

routiful, 2018-05-15:
Thank you for sharing.

Can you add one more wire to transfer power(24v) to daisy-chain (Roll1, pitch, Roll2)?

routiful, 2018-05-16:
Dynamixel-workbench is based on Dynamixel SDK. Before I tested it few days ago with dynamixel_workbench_controllers position controller, it is moved right position. (BaudRate : 2000000) 

How about to download latest version of open_manipulator and dynamixel_workbench in master branch and try again?

robotpilot, 2018-05-22:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

gokul-gokz, 2018-05-23:
I implemented the latest version of the open manipulator. But still it throws the same error.

robotpilot, 2018-05-28:
@routiful, Please check back on this issue.

routiful, 2018-05-29:
Hi, sorry to late reply.

I had tested  Manipulator-H is assembled with three H54-200-S500-R(id : 1, 11, 12),  two H54-100-S500-R(id : 3,4) and two H42-20-S300-R(id : 5, 6), every Dynamixel Pro is set 2000000 Baud Rate, and i used open_manipulator package ([tag 0.1.1](https://github.com/ROBOTIS-GIT/open_manipulator/tree/0.1.1)) 

I attached edited launch file(dynamixel_controller.launch)

```
<launch>
  <arg name="device_name"          default="/dev/ttyUSB0"/>
  <arg name="baud_rate"            default="2000000"/>

  <arg name="scan_range"           default="20"/>

  <param name="device_name"        value="$(arg device_name)"/>
  <param name="baud_rate"          value="$(arg baud_rate)"/>

  <param name="scan_range"         value="$(arg scan_range)"/>

  <node name="open_manipulator_dynamixel_controller" pkg="open_manipulator_dynamixel_ctrl" type="dynamixel_controller" required="true" output="screen"/>
</launch>

```
I checked topic (/robotis/dynamixel/present_states) after launching it.
```
---
header: 
  seq: 383
  stamp: 
    secs: 1527562798
    nsecs: 922348601
  frame_id: ''
name: [id_1, id_3, id_4, id_5, id_6, id_11, id_12]
position: [-1.8249231576919556, 0.37079060077667236, 0.0017275186255574226, 0.5331647992134094, -0.10352593660354614, 0.06449402868747711, 0.2680032253265381]
velocity: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
effort: []
---
```

I can't find any problem to use package. Would you try one more to follow above instruction?

Thanks