ROBOTIS-GIT/open_manipulator #38: Where are the CAD files for open_manipulator?

dshinsel, 2018-08-18:
I see a post a *YEAR AGO* saying the STL files would be released, but still can't find a way to download the STL files?  Can you please post them someplace (NOT OnShape!), where we can download the files?  OnShape has the files locked   :-(

If this is an *OPEN* project, please make the files openly available!

MyNameIsCosmo, 2018-08-18:
@dshinsel 
I was able to freely access the OnShape cad document and export the files.

Out of curiosity, what is your issue accessing the following files?
https://github.com/ROBOTIS-GIT/open_manipulator/tree/master/open_manipulator_description/meshes

routiful, 2018-08-19:
@dshinsel 

I think you have not searched how to download cad files on OnShape.
I kindly give you a reference how to download it. 

https://cad.onshape.com/help/Content/exporting-files.htm

http://emanual.robotis.com/docs/en/platform/turtlebot3/faq/#how-to-download-the-stl-files-of-turtlebot3

If you have any further questions, please open an issue any time

Thanks very much.

MONICA-PALLA, 2018-08-20:
**@routiful** 
May I know the material used for 3D printing of various parts? 

Thank you,
Monica, P.

routiful, 2018-08-20:
@MONICA-PALLA 
We tested ABS and PLA. In my experiences, the ABS is better.

@dshinsel  
Yes, you have to login.

MONICA-PALLA, 2018-08-21:
**@routiful** 
Thank you very much !

MyNameIsCosmo, 2018-08-22:
I think Dave's concern is easily accessing 3d-printable meshes of the CAD files.
A website such as Thingiverse would be great to not only host the STL files, but also bring more attention to this project as a Thingiverse is a hub for do-it-yourself manufacturing.

Here are some Thingiverse projects that are related to the original Turtlebot platform for example:
https://www.thingiverse.com/thing:11193
https://www.thingiverse.com/thing:24012

If the hardware design is versioned (v1, v2, etc), the 3d mesh files hosted on Thingiverse could either be easily updated or a new project could be created to reflect these changes.

robotpilot, 2018-08-22:
Hi @MyNameIsCosmo, 
I also feel that Thingiverse is a great feature and popular among a lot of users. In addition to releasing the project to Onshape, I will consider sharing the file separately or uploading the file to the Thingiverse.
Thank you for your consideration and supports. :)

dshinsel, 2018-08-23:
@robotpilot - Hi Pyo!  Great to hear from you!
I agree with @MyNameIsCosmo, Thingiverse would be really great. I think there are two target audiences:
1.  Someone that just wants to build the open_manipulator hardware as specified (Thingiverse is best for that; requiring someone to get an OnShape account for that is frustrating)

2. Someone that wants to modify / customize the hardware; I think requiring a person to get an OnShape account for editing makes perfect sense!

robotpilot, 2018-08-23:
@dshinsel Hi Dave,
I also agree with the opinion of @MyNameIsCosmo and you. 
We will make it easier to download, use and modify.
Thank you so much! :)

robotpilot, 2018-09-03:
We have released the OpenManipulator CAD file at the link below.

https://www.thingiverse.com/thing:3069574

The TurtleBot3 and other open-source hardware were also released. Feel free to use it. :)

https://www.thingiverse.com/ROBOTIS/designs

robotpilot, 2018-09-03:
Thanks, @dshinsel and @MyNameIsCosmo
With your valuable feedback, the now and upcoming open source hardware has also been added to Thingiverse. This issue has been resolved and will be closed. Thank you.  :)

closed 2018-09-03 by robotpilot