ROBOTIS-GIT/open_manipulator #39: About open_manipulator_dynamixel_controller

YunchuZhang, 2018-08-22:
Hi dear,
I try to use the open_manipulator_dynamixel_controller package in ros. But, when I run it , it shows that 
[ INFO] [1534897770.150706237]: open_manipulator_dynamixel_controller : Init OK!
Segmentation fault (core dumped)

But I do not change the code. Would you like to help me figure it out?
Thanks

YunchuZhang, 2018-08-22:
@robotpilot  Hi my friends, this problem always exist when I try a new desktop. If I want to control real dynamixel motors through moveit, which package do I need to use?

Thanks

routiful, 2018-08-22:
Hi :)

Can I know how to set your development environments? and launch file?

https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_dynamixel_ctrl/launch/dynamixel_controller.launch

YunchuZhang, 2018-08-22:
rosrun open_manipulator_dynamixel_ctrl dynamixel_controller 
I run the cpp directly,and tune motor's ID to 12,13,14,15 which could be found in your windows software

YunchuZhang, 2018-08-22:
my development environment is Ubuntu 16.04 with ros-kinetic and dynamixel xl-320 motors

routiful, 2018-08-23:
XL-320 does not support `current_mode`

please modifiy `current_mode` to `position_mode`.

https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_dynamixel_ctrl/launch/dynamixel_controller.launch#L14

YunchuZhang, 2018-08-23:
Thanks, but when I delete the gripper part, and only use 4 xl-320 motors for the arm like this:

  <arg name="joint_controller"       default="position_mode"/>

  <arg name="joint1_id"              default="12"/>
  <arg name="joint2_id"              default="13"/>
  <arg name="joint3_id"              default="14"/>
  <arg name="joint4_id"              default="15"/>

it still shows that problem:[ INFO] [1534897770.150706237]: open_manipulator_dynamixel_controller : Init OK!
Segmentation fault (core dumped)

By the way, does xl-430 support current_node?

YunchuZhang, 2018-08-24:
@routiful also, the subsequent question is that, how can I control the real robot motor with moveit since I do not find action-service in your packages.

Thanks

YunchuZhang, 2018-08-27:
So when I change  current mode to position_mode, both 320 and 430 will support that code? Does AX-18 support current_node?

Also, as for those topic:
  /open_manipulator/get_joint_position
  /open_manipulator/get_kinematics_pose
  /open_manipulator/set_gripper_position
  /open_manipulator/set_joint_position
  /open_manipulator/set_kinematics_pose

I see them in open_manipulator_position_ctrl package, what is the way that I can use them? To rosrun the moveit and this together? And I will not need to write the action service that receive move_group message from moveit?

Thanks

routiful, 2018-08-27:
1. You are supposed to change(not DELETE) code from `current_mode` to `position_mode`. 
Because OpenManipulator mainly support XM430-W350, this launch file has been set `current_mode` to gripper controller. And AX-18 does not support current_mode. 

You can find which Dynamixels supports current_mode in e-manual

http://emanual.robotis.com/docs/en/dxl/ax/ax-12w/#cwccw-angle-limit6-8

http://emanual.robotis.com/docs/en/dxl/x/xm430-w350/#operating-mode11

2. Bringup OpenManipulator and launch open_manipulator_demo node. Below manual and lecture might be help you.

http://emanual.robotis.com/docs/en/platform/openmanipulator/#bringup
http://emanual.robotis.com/docs/en/platform/openmanipulator/#manipulation

https://youtu.be/dekcltSZkNE?t=30m6s

3. open_manipulator_demo node includes moveit API. So you don't need to send action service to moveit. (But you can) The provided topics are going to generate moveit API to control manipulator.

YunchuZhang, 2018-08-27:
Thanks, but when I change it to position node,and use the launch file I tuned, it still has bug:

NODES
  /
    dynamixel_controller (open_manipulator_dynamixel_ctrl/dynamixel_controller)

ROS_MASTER_URI=http://localhost:11311

process[dynamixel_controller-1]: started with pid [19213]
[ INFO] [1535402487.454517542]: open_manipulator_dynamixel_controller : Init OK!
================================================================================REQUIRED process [dynamixel_controller-1] has died!
process has died [pid 19213, exit code -11, cmd /home/yunchu/catkin/devel/lib/open_manipulator_dynamixel_ctrl/dynamixel_controller __name:=dynamixel_controller __log:=/home/yunchu/.ros/log/aa1a442c-a66c-11e8-904c-90324b8e481b/dynamixel_controller-1.log].
log file: /home/yunchu/.ros/log/aa1a442c-a66c-11e8-904c-90324b8e481b/dynamixel_controller-1*.log
Initiating shutdown!
================================================================================
[dynamixel_controller-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

This is my launch file:
<launch>
  <arg name="use_robot_name"         default="open_manipulator"/>
  <arg name="device_name"            default="/dev/ttyUSB0"/>
  <arg name="baud_rate"              default="1000000"/>
  <arg name="protocol_version"       default="2.0"/>

  <arg name="joint_controller"       default="position_mode"/>

  <arg name="joint1_id"              default="12"/>
  <arg name="joint2_id"              default="13"/>
  <arg name="joint3_id"              default="14"/>
  <arg name="joint4_id"              default="15"/>

  <arg name="gripper_controller"     default="position_mode"/>

  <arg name="gripper_id"             default="11"/>

  <node pkg="open_manipulator_dynamixel_ctrl" type="dynamixel_controller" name="dynamixel_controller" required="true" output="screen">
    <param name="robot_name"           value="$(arg use_robot_name)"/>
    <param name="device_name"          value="$(arg device_name)"/>
    <param name="baud_rate"            value="$(arg baud_rate)"/>
    <param name="protocol_version"     value="$(arg protocol_version)"/>

    <param name="joint_controller"     value="$(arg joint_controller)"/>

    <param name="joint1_id"            value="$(arg joint1_id)"/>
    <param name="joint2_id"            value="$(arg joint2_id)"/>
    <param name="joint3_id"            value="$(arg joint3_id)"/>
    <param name="joint4_id"            value="$(arg joint4_id)"/>

    <param name="gripper_controller"   value="$(arg gripper_controller)"/>

    <param name="gripper_id"           value="$(arg gripper_id)"/>
  </node>
</launch>

routiful, 2018-08-29:
Sorry, can I get your launch file?

YunchuZhang, 2018-08-29:
This is my launch file

[dynamixel_controller.launch.tar.gz](https://github.com/ROBOTIS-GIT/open_manipulator/files/2333725/dynamixel_controller.launch.tar.gz)

YunchuZhang, 2018-08-30:
@routiful @robotpilot 
Today I try to use 5 new AX-18 motors, but it still has problems:

SUMMARY
========

PARAMETERS
 * /dynamixel_controller/baud_rate: 1000000
 * /dynamixel_controller/device_name: /dev/ttyUSB0
 * /dynamixel_controller/gripper_controller: position_mode
 * /dynamixel_controller/gripper_id: 5
 * /dynamixel_controller/joint1_id: 1
 * /dynamixel_controller/joint2_id: 2
 * /dynamixel_controller/joint3_id: 3
 * /dynamixel_controller/joint4_id: 4
 * /dynamixel_controller/joint_controller: position_mode
 * /dynamixel_controller/protocol_version: 1.0
 * /dynamixel_controller/robot_name: open_manipulator
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    dynamixel_controller (open_manipulator_dynamixel_ctrl/dynamixel_controller)

ROS_MASTER_URI=http://localhost:11311

process[dynamixel_controller-1]: started with pid [11606]
[ INFO] [1535670929.616498968]: open_manipulator_dynamixel_controller : Init OK!
================================================================================REQUIRED process [dynamixel_controller-1] has died!
process has died [pid 11606, exit code -11, cmd /home/yunchu/catkin/devel/lib/open_manipulator_dynamixel_ctrl/dynamixel_controller __name:=dynamixel_controller __log:=/home/yunchu/.ros/log/b3641f10-aca9-11e8-82e3-90324b8e481b/dynamixel_controller-1.log].
log file: /home/yunchu/.ros/log/b3641f10-aca9-11e8-82e3-90324b8e481b/dynamixel_controller-1*.log
Initiating shutdown!
================================================================================
[dynamixel_controller-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

routiful, 2018-08-31:
Sorry for late reply.

I solved bug about protocol 1.0 problems.
You can download below commands.(Before you change repository, please purge yours)

```
$ git clone -b develop https://github.com/ROBOTIS-GIT/open_manipulator.git
``

YunchuZhang, 2018-08-31:
thanks, does it contains the `` on the end "$ git clone -b develop https://github.com/ROBOTIS-GIT/open_manipulator.git`` "

Also, if I want to use 5 xl430 motors, does the problem be solved?
Thanks

routiful, 2018-09-07:
Sorry for late reply, did you solve xl430 problem?
I tested five XL430-W250-T(http://emanual.robotis.com/docs/en/dxl/x/xl430-w250/) and it worked fine.

YunchuZhang, 2018-09-07:
not yet,since i do not have 5 xl430 motors on my side. Do you test the code from here git clone -b develop https://github.com/ROBOTIS-GIT/open_manipulator.git
and change protocol to 2.0? 
thanks