ROBOTIS-GIT/open_manipulator #51: Distinct world coordinates?

Affonso-Gui, 2018-11-15:
Hello all,

Is there a reason to have different world coordinates for the urdf ([open_manipulator.urdf.xacro](https://github.com/ROBOTIS-GIT/open_manipulator/blob/feature_robotis_manipulator/open_manipulator_description/urdf/open_manipulator.urdf.xacro#L59)) and om_chain ([om_chain.cpp](https://github.com/ROBOTIS-GIT/open_manipulator/blob/feature_robotis_manipulator/open_manipulator_libs/src/om_chain.cpp#L36)) ?

I understand that in the om_chain the X axis origin is aligned to the manipulator in the zero angle position, but wouldn't it be better to unify with the urdf origin? Having the same world coordinates should also abbreviate some coordinate transforms when *e.g.* getting the target object coordinates from vision -> solving the IK through `goal_task_space_path` service.

[image]

gpwhd07, 2018-11-21:
Hi, @Affonso-Gui 
Thank you for your contribution.

We also knew about that matter.
Therefore, we set the coordinate system in om_chain.cpp was matched with the URDF coordinate system.
For the changed code, it is uploaded to the open branch of open_manipulator.

From now on, we will do all work in open_manipulator, open_manipulator_msgs, open_manipulator_simulations, robotis_manipulator package with develop branch before releasing (upload to master branch).
Please refer to your work.

Affonso-Gui, 2018-11-21:
Confirmed that the origins match :+1: 

Roger about the develop branch!

closed 2018-11-21 by Affonso-Gui