ROBOTIS-GIT/open_manipulator #54: Request to add the "RAIL BLOCK" stl file to Thingiverse and a question about the "part lists" section in the e-manual

mjyc, 2018-11-19:
Thanks @routiful for [your quick response](https://discourse.ros.org/t/introducing-the-openmanipulator/2595/9) and updating 3d part model names in [Thingsverse](https://www.thingiverse.com/thing:3069574/files) already!

I started to work on it and noticed the "RAIL BLOCK" part mentioned in "Assembly Manual" is missing. Can you add this part to Thingiverse as well?

One quick question: I also noticed the ["part lists" section in the e-manual](http://emanual.robotis.com/docs/en/platform/openmanipulator/) is still not matching with the part list in the Assembly Manual. Based on @routiful 's 3rd response item at [discourse.ros](https://discourse.ros.org/t/introducing-the-openmanipulator/2595/9) (`The spreadsheet will be deleted, I have a plan to move it in e-manual.`), Assembly Manual is the most up to date document, so I'll continue working with the Assembly Manual, ignoring additional parts mentioned in the e-manual.

Ryan-JHShim, 2018-11-20:
Hi mjyc, 

Thank you for your comments. We just updated our thingiverse page. (https://www.thingiverse.com/thing:3069574) 

Also, we updated our assembly manual most recently, so that is what you should refer to. We will update all the other item lists within a month. 

Cheers,
Ryan

mjyc, 2018-11-20:
Thanks for the clarification!

closed 2018-11-20 by mjyc