ROBOTIS-GIT/open_manipulator #58: [devel] Do not disable torque on shutdown

Affonso-Gui, 2018-11-21:
Sometimes disabling the torque fails on `open_manipulator_controller` shutdown:

```
[INFO] Shutdown the OpenManipulator
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

Any ideas on what may be causing this?
@t-kitajima

t-kitajima, 2018-11-21:
I tried debugging with gdb.
It looks like SIGSEGV with Protocol2PacketHandler of DynamixelSDK.
I attach a debug log.

```[New Thread 0x7ffff0cc0700 (LWP 22463)]
[New Thread 0x7fffebfff700 (LWP 22464)]
[New Thread 0x7fffeb7fe700 (LWP 22465)]
[New Thread 0x7fffeaffd700 (LWP 22470)]
Joint Dynamixel ID : 11, Model Name : XM430-W350
Joint Dynamixel ID : 12, Model Name : XM430-W350
Joint Dynamixel ID : 13, Model Name : XM430-W350
Joint Dynamixel ID : 14, Model Name : XM430-W350
[ INFO] [1542792566.756314946]: id=11, onoff=0
[ INFO] [1542792566.764126267]: id=11, onoff=1
[ INFO] [1542792566.766125774]: id=12, onoff=0
[ INFO] [1542792566.774124079]: id=12, onoff=1
[ INFO] [1542792566.776123747]: id=13, onoff=0
[ INFO] [1542792566.784124909]: id=13, onoff=1
[ INFO] [1542792566.786127109]: id=14, onoff=0
[ INFO] [1542792566.794131189]: id=14, onoff=1
Gripper Dynamixel ID : 15, Model Name : XM430-W350
[ INFO] [1542792566.832212999]: id=15, onoff=0
[ INFO] [1542792566.838173777]: id=15, onoff=1
[ INFO] [1542792566.844152138]: id=11, onoff=1
[ INFO] [1542792566.846129929]: id=12, onoff=1
[ INFO] [1542792566.848128708]: id=13, onoff=1
[ INFO] [1542792566.850132817]: id=14, onoff=1
[ INFO] [1542792566.852136259]: id=15, onoff=1
[New Thread 0x7fffea7fc700 (LWP 22488)]
[ INFO] [1542792566.862433965]: OpenManipulator initialization
[ INFO] [1542792566.862457696]: OpenManipulator control loop start
[ INFO] [1542792571.852622923]: id=11, onoff=0
[DynamixelWorkbench] Failed to change torque status!
[ INFO] [1542792571.852686758]: id=15, onoff=0

Thread 1 "open_manipulato" received signal SIGSEGV, Segmentation fault.
0x00007ffff3e4900b in dynamixel::Protocol2PacketHandler::rxPacket (this=0x687c40, port=0x699120, rxpacket=0x7fffffffc3d0 "")
    at /home/kitajima/om_ws/src/DynamixelSDK/ros/src/dynamixel_sdk/protocol2_packet_handler.cpp:312
312	        if ((rxpacket[idx] == 0xFF) && (rxpacket[idx+1] == 0xFF) && (rxpacket[idx+2] == 0xFD) && (rxpacket[idx+3] != 0xFD))
```

gpwhd07, 2018-11-23:
Hi, @Affonso-Gui and @t-kitajima 
I tested about this shutdown issue.
In my system, it was successfully worked. 

> robotis@spc:~$ roslaunch open_manipulator_controller open_manipulator_controller.launch
> ... logging to /home/robotis/.ros/log/011c753e-eed8-11e8-bdd5-ac9e17829ad5/roslaunch-spc-6281.log
> Checking log directory for disk usage. This may take awhile.
> Press Ctrl-C to interrupt
> Done checking log file disk usage. Usage is <1GB.
> 
> started roslaunch server http://localhost:32967/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /open_manipulator_controller/baud_rate: 1000000
>  * /open_manipulator_controller/robot_name: open_manipulator
>  * /open_manipulator_controller/usb_port: /dev/ttyUSB0
>  * /open_manipulator_controller/using_platform: True
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.14
> 
> NODES
>   /
>     open_manipulator_controller (open_manipulator_controller/open_manipulator_controller)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> process[open_manipulator_controller-1]: started with pid [6304]
> Joint Dynamixel ID : 11, Model Name : XM430-W350
> Joint Dynamixel ID : 12, Model Name : XM430-W350
> Joint Dynamixel ID : 13, Model Name : XM430-W350
> Joint Dynamixel ID : 14, Model Name : XM430-W350
> Gripper Dynamixel ID : 15, Model Name :XM430-W350
> [INFO] Start the OpenManipulator control thread
> [INFO] Successed to OpenManipulator initialization
> ^C[open_manipulator_controller-1] killing on exit
> [INFO] Shutdown the OpenManipulator
> shutting down processing monitor...
> ... shutting down processing monitor complete
> done

`[ERROR] [Dynamixel Workbench] Failed to change torque status!` occurs when controller couldn't get a response from dxls, I think that the cable connections have problem.

Please, retest with the new source code of the 'develop branch', and let me know more detail about your test configuration.

Affonso-Gui, 2018-11-29:
Tried to update the sources, but the problem still prevails.

I am using the `develop` branch of:
- dynamixel-workbench
- dynamixel-workbench-msgs
- open_manipulator
- open_manipulator_msgs
- robotis_manipulator

And version 3.6.2 of dynamixel_sdk (also tried with source from master branch, without changes).

I am not getting error messages on shutdown anymore, but many times the torque is left enabled. When calling the `/set_torque_state` service, I get the following error every time:

```
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[open_manipulator_controller-2] process has died [pid 13116, exit code -11, cmd /home/robotis/manipulator_ws/devel/lib/open_manipulator_controller/open_manipulator_controller __name:=open_manipulator_controller __log:=/home/robotis/.ros/log/8c07661e-f382-11e8-aba5-4ccc6a7f3222/open_manipulator_controller-2.log].
log file: /home/robotis/.ros/log/8c07661e-f382-11e8-aba5-4ccc6a7f3222/open_manipulator_controller-2*.log
```
I am also having the following warning, which disappears when reducing the frequency of `ACTUATOR_CONTROL_TIME` `ITERATION_FREQUENCY` `ACTUATOR_CONTROL_TIME_MSEC`.

```
[WARN] Over the control time : 0.032
```

Full launch text with XL model:
```
$ roslaunch open_manipulator_controller open_manipulator_controller.launch

... logging to /home/robotis/.ros/log/8c07661e-f382-11e8-aba5-4ccc6a7f3222/roslaunch-ROBOTIS-JAPAN03-13086.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ROBOTIS-JAPAN03:34350/

SUMMARY
========

PARAMETERS
 * /open_manipulator_controller/baud_rate: 1000000
 * /open_manipulator_controller/robot_name: open_manipulator
 * /open_manipulator_controller/usb_port: /dev/ttyUSB0
 * /open_manipulator_controller/using_platform: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    open_manipulator_controller (open_manipulator_controller/open_manipulator_controller)

auto-starting new master
process[master]: started with pid [13096]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 8c07661e-f382-11e8-aba5-4ccc6a7f3222
process[rosout-1]: started with pid [13109]
started core service [/rosout]
process[open_manipulator_controller-2]: started with pid [13116]
Joint Dynamixel ID : 11, Model Name : XL430-W250
Joint Dynamixel ID : 12, Model Name : XL430-W250
Joint Dynamixel ID : 13, Model Name : XL430-W250
Joint Dynamixel ID : 14, Model Name : XL430-W250
Gripper Dynamixel ID : 15, Model Name :XL430-W250
[ERROR] [DynamixelWorkbench] Failed to set Current Based Position Control Mode!
[INFO] Start the OpenManipulator control thread
[INFO] Successed to OpenManipulator initialization
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[open_manipulator_controller-2] process has died [pid 13116, exit code -11, cmd /home/robotis/manipulator_ws/devel/lib/open_manipulator_controller/open_manipulator_controller __name:=open_manipulator_controller __log:=/home/robotis/.ros/log/8c07661e-f382-11e8-aba5-4ccc6a7f3222/open_manipulator_controller-2.log].
log file: /home/robotis/.ros/log/8c07661e-f382-11e8-aba5-4ccc6a7f3222/open_manipulator_controller-2*.log
```

And with XM model:
```
$ roslaunch open_manipulator_controller open_manipulator_controller.launch 
... logging to /home/robotis/.ros/log/537f3a5a-f383-11e8-aba5-4ccc6a7f3222/roslaunch-ROBOTIS-JAPAN03-13282.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://ROBOTIS-JAPAN03:40445/

SUMMARY
========

PARAMETERS
 * /open_manipulator_controller/baud_rate: 1000000
 * /open_manipulator_controller/robot_name: open_manipulator
 * /open_manipulator_controller/usb_port: /dev/ttyUSB0
 * /open_manipulator_controller/using_platform: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    open_manipulator_controller (open_manipulator_controller/open_manipulator_controller)

auto-starting new master
process[master]: started with pid [13292]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 537f3a5a-f383-11e8-aba5-4ccc6a7f3222
process[rosout-1]: started with pid [13305]
started core service [/rosout]
process[open_manipulator_controller-2]: started with pid [13312]
Joint Dynamixel ID : 11, Model Name : XM430-W350
Joint Dynamixel ID : 12, Model Name : XM430-W350
Joint Dynamixel ID : 13, Model Name : XM430-W350
Joint Dynamixel ID : 14, Model Name : XM430-W350
Gripper Dynamixel ID : 15, Model Name :XM430-W350
[INFO] Start the OpenManipulator control thread
[INFO] Successed to OpenManipulator initialization
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[ERROR] [DynamixelWorkbench] Failed to change torque status!
[open_manipulator_controller-2] process has died [pid 13312, exit code -11, cmd /home/robotis/manipulator_ws/devel/lib/open_manipulator_controller/open_manipulator_controller __name:=open_manipulator_controller __log:=/home/robotis/.ros/log/537f3a5a-f383-11e8-aba5-4ccc6a7f3222/open_manipulator_controller-2.log].
log file: /home/robotis/.ros/log/537f3a5a-f383-11e8-aba5-4ccc6a7f3222/open_manipulator_controller-2*.log
```
OS infomation:
```
$ uname -a
Linux ROBOTIS-JAPAN03 4.4.0-139-generic #165-Ubuntu SMP Wed Oct 24 10:58:50 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```