ROBOTIS-GIT/open_manipulator #61: Build Error Question  about master brunch version 1.0.0

hyunoklee, 2018-11-29:
Open manipulator master git is updated version 1.0.0 9days ago. 
It seems that motion accuracy has improved tremendously. Very Wonderful!. 

But there is some build issue.  
All git statues is master brunch .  
I use below command to download.   
$git clone https://github.com/ROBOTIS-GIT/open_manipulator_simulations.git  
$git clone https://github.com/ROBOTIS-GIT/open_manipulator.git  
$git clone https://github.com/ROBOTIS-GIT/open_manipulator_msgs.git
$git clone https://github.com/ROBOTIS-GIT/robotis_manipulator.git
$git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench.git  
$git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench-msgs.git  
$git clone https://github.com/ROBOTIS-GIT/DynamixelSDK.git

When I build code.
I can see Erro Log about dynamixel-workbench. 
Others have written naver cafe about the same build issue.
https://cafe.naver.com/openrt/19443

So I use back up code, the code is hackerton-beta-brunch dynamixel-workbench code. 
git clone https://github.com/hyunoklee/dynamixel-workbench.git
git clone https://github.com/hyunoklee/dynamixel-workbench-msgs.git

please check about that.  
Thank you.  

//////////////////////////ERRO LOG///////////////////////

/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp: In member function ‘virtual void DYNAMIXEL::JointDynamixel::enable()’:
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:70:36: error: ‘class DynamixelWorkbench’ has no member named ‘torqueOn’
     result = dynamixel_workbench_->torqueOn(dynamixel_.id.at(index), &log);
                                    ^
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp: In member function ‘virtual void DYNAMIXEL::JointDynamixel::disable()’:
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:85:36: error: ‘class DynamixelWorkbench’ has no member named ‘torqueOff’
     result = dynamixel_workbench_->torqueOff(dynamixel_.id.at(index), &log);
                                    ^
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp: In member function ‘bool DYNAMIXEL::JointDynamixel::initialize(std::vector<unsigned char>, STRING, STRING)’:
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:127:34: error: ‘class DynamixelWorkbench’ has no member named ‘init’
   result = dynamixel_workbench_->init(dxl_device_name.c_str(), std::atoi(dxl_baud_rate.c_str()), &log);
                                  ^
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:137:68: error: no matching function for call to ‘DynamixelWorkbench::ping(uint8_t&, uint16_t*, const char**)’
     result = dynamixel_workbench_->ping(id, &get_model_number, &log);
                                                                    ^
In file included from /home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/../include/open_manipulator_libs/Dynamixel.h:27:0,
                 from /home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:19:
/home/hyunoklee/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:49:8: note: candidate: bool DynamixelWorkbench::ping(uint8_t, uint16_t*)
   bool ping(uint8_t id, uint16_t *get_model_number = 0);
        ^
/home/hyunoklee/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:49:8: note:   candidate expects 2 arguments, 3 provided
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp: In member function ‘bool DYNAMIXEL::JointDynamixel::setOperatingMode(std::vector<unsigned char>, STRING)’:
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:168:91: error: no matching function for call to ‘DynamixelWorkbench::jointMode(__gnu_cxx::__alloc_traits<std::allocator<unsigned char> >::value_type&, const uint32_t&, const uint32_t&, const char**)’
       result = dynamixel_workbench_->jointMode(actuator_id.at(num), velocity, effort, &log);
                                                                                           ^
In file included from /home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/../include/open_manipulator_libs/Dynamixel.h:27:0,
                 from /home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:19:
/home/hyunoklee/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:64:8: note: candidate: bool DynamixelWorkbench::jointMode(uint8_t, uint16_t, uint16_t)
   bool jointMode(uint8_t id, uint16_t vel = 0, uint16_t acc = 0);
        ^
/home/hyunoklee/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:64:8: note:   candidate expects 3 arguments, 4 provided
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:179:38: error: ‘class DynamixelWorkbench’ has no member named ‘CurrentBasedPositionMode’
       result = dynamixel_workbench_->CurrentBasedPositionMode(actuator_id.at(num), current, &log);
                                      ^
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:190:91: error: no matching function for call to ‘DynamixelWorkbench::jointMode(__gnu_cxx::__alloc_traits<std::allocator<unsigned char> >::value_type&, const uint32_t&, const uint32_t&, const char**)’
       result = dynamixel_workbench_->jointMode(actuator_id.at(num), velocity, effort, &log);
                                                                                           ^
In file included from /home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/../include/open_manipulator_libs/Dynamixel.h:27:0,
                 from /home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:19:
/home/hyunoklee/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:64:8: note: candidate: bool DynamixelWorkbench::jointMode(uint8_t, uint16_t, uint16_t)
   bool jointMode(uint8_t id, uint16_t vel = 0, uint16_t acc = 0);
        ^
[... 112 lines cut ...]
/home/hyunoklee/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:74:8: note: candidate: bool DynamixelWorkbench::syncWrite(uint8_t*, uint8_t, const char*, int32_t*)
   bool syncWrite(uint8_t *id, uint8_t id_num, const char *item_name, int32_t *data);
        ^
/home/hyunoklee/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:74:8: note:   candidate expects 4 arguments, 3 provided
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp: In member function ‘double DYNAMIXEL::GripperDynamixel::receiveDynamixelValue()’:
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:567:47: error: no matching function for call to ‘DynamixelWorkbench::syncRead(int, uint8_t [1], uint8_t, const char**)’
                                           &log);
                                               ^
In file included from /home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/../include/open_manipulator_libs/Dynamixel.h:27:0,
                 from /home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:19:
/home/hyunoklee/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:79:12: note: candidate: int32_t* DynamixelWorkbench::syncRead(const char*)
   int32_t* syncRead(const char* item_name);              // sync read
            ^
/home/hyunoklee/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:79:12: note:   candidate expects 1 argument, 4 provided
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:573:34: error: ‘class DynamixelWorkbench’ has no member named ‘getSyncReadData’
   result = dynamixel_workbench_->getSyncReadData(SYNC_READ_HANDLER_FOR_PRESENT_POSITION_VELOCITY_CURRENT, 
                                  ^
[... cut ...]
open_manipulator/open_manipulator_libs/CMakeFiles/open_manipulator_libs.dir/build.make:110: recipe for target 'open_manipulator/open_manipulator_libs/CMakeFiles/open_manipulator_libs.dir/src/Dynamixel.cpp.o' failed
make[2]: *** [open_manipulator/open_manipulator_libs/CMakeFiles/open_manipulator_libs.dir/src/Dynamixel.cpp.o] Error 1
CMakeFiles/Makefile2:4212: recipe for target 'open_manipulator/open_manipulator_libs/CMakeFiles/open_manipulator_libs.dir/all' failed
make[1]: *** [open_manipulator/open_manipulator_libs/CMakeFiles/open_manipulator_libs.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

routiful, 2018-11-30:
Sorry for inconvenience. 
Please pull master branch and make again.

hyunoklee, 2018-11-30:
Sorry . 
I did not build and closed the issue.
Now I can see another build error.
So I reopen the issue.
I attach build error log below.  please check~
////////build error log////////////

_manager_gui/moc_main_window.cpp.o
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp: In member function ‘bool DYNAMIXEL::JointDynamixel::setOperatingMode(std::vector<unsigned char>, STRING)’:
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:179:38: error: ‘class DynamixelWorkbench’ has no member named ‘CurrentBasedPositionMode’
       result = dynamixel_workbench_->CurrentBasedPositionMode(actuator_id.at(num), current, &log);
                                      ^
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp: In member function ‘std::vector<ROBOTIS_MANIPULATOR::_Point> DYNAMIXEL::JointDynamixel::receiveAllDynamixelValue(std::vector<unsigned char>)’:
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:309:117: error: no matching function for call to ‘DynamixelWorkbench::convertValue2Current(__gnu_cxx::__alloc_traits<std::allocator<unsigned char> >::value_type&, uint16_t)’
       actuator.effort = dynamixel_workbench_->convertValue2Current(actuator_id.at(index), (uint16_t)get_value[index]);
                                                                                                                     ^
In file included from /home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/../include/open_manipulator_libs/Dynamixel.h:27:0,
                 from /home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:19:
/home/hyunoklee/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:90:9: note: candidate: float DynamixelWorkbench::convertValue2Current(int16_t)
   float convertValue2Current(int16_t value);
         ^
/home/hyunoklee/catkin_ws/src/dynamixel-workbench/dynamixel_workbench_toolbox/include/dynamixel_workbench_toolbox/dynamixel_workbench.h:90:9: note:   candidate expects 1 argument, 2 provided
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp: In member function ‘bool DYNAMIXEL::GripperDynamixel::setOperatingMode(STRING)’:
/home/hyunoklee/catkin_ws/src/open_manipulator/open_manipulator_libs/src/Dynamixel.cpp:482:36: error: ‘class DynamixelWorkbench’ has no member named ‘CurrentBasedPositionMode’
     result = dynamixel_workbench_->CurrentBasedPositionMode(dynamixel_.id.at(0), current, &log);

routiful, 2018-12-03:
We are updated dynamixel-workbench and openmanipulator repo.
Please pull every repository including msgs.

hyunoklee, 2018-12-04:
thank you
build ok

closed 2018-12-04 by hyunoklee