ROBOTIS-GIT/open_manipulator #62: Travis failing

Affonso-Gui, 2018-11-30:
Travis fails with `robotis_manipulator` dependency problems:

```
ERROR: the following packages/stacks could not have their rosdep keys resolved to system dependencies:

open_manipulator_libs: Cannot locate rosdep definition for [robotis_manipulator]
open_manipulator_controller: Cannot locate rosdep definition for [robotis_manipulator]

ici_time:end:156b94b55321a25c:start=1543490803809231452,finish=1543490805488577586,duration=1679346134[0K
ici_fold:end:rosdep_install<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<

Function rosdep_install returned with code '1' after 0 min 1 sec

travis_time:end:1f92e828:start=1543490721179475105,finish=1543490805707014621,duration=84527539516
[0K[31;1mThe command "source .ci_config/travis.sh" exited with 1.[0m

Done. Your build exited with 1.
```

Affonso-Gui, 2018-11-30:
This problem seems to be happening since https://github.com/ROBOTIS-GIT/open_manipulator/commit/b007fd81a2b5dd5b227774595327791953a59c01 (update that uses `robotis_manipulator`).

Affonso-Gui, 2018-11-30:
Thanks for your reply.

Please let me know when it is updated, so we can check and close this issue.

gpwhd07, 2018-12-12:
Hi, @Affonso-Gui 

We uploaded all new OpenManipulator dependency pkg to master branch. Please check that. It was successfully worked in our enviroments 

And I'm going to close this issue until you update the issue with more information. If you do so I'll reopen the issue.

Affonso-Gui, 2018-12-20:
It appears that the travis tests continues to fail for all comits after `update 1.0.0` https://github.com/ROBOTIS-GIT/open_manipulator/commit/b007fd81a2b5dd5b227774595327791953a59c01

@gpwhd07 Please reopen this issue

Affonso-Gui, 2019-01-29:
@gpwhd07 This is not solved. Please reopen

Affonso-Gui, 2019-02-13:
Seems to be passing again in the latest tests.

https://travis-ci.org/ROBOTIS-GIT/open_manipulator/builds/490374120?utm_source=github_status&utm_medium=notification
https://travis-ci.org/ROBOTIS-GIT/open_manipulator/builds/490403940?utm_source=github_status&utm_medium=notification

Maybe these changes made the difference:
https://github.com/ROBOTIS-GIT/open_manipulator/commit/a171b4b91d988bbbc9f41f1cccdaf0d4b28e0c9f#diff-354f30a63fb0907d4ad57269548329e3

Closing this for now.

closed 2019-02-13 by Affonso-Gui