ROBOTIS-GIT/open_manipulator #64: [checkLimit] question

hyunoklee, 2018-12-04:
I do not know if this is an issue. Just refer to it. 
CheckLimit function  is added at this time updated open manipulator code.  
At start time, There is no problem if Open Manipulator's joint position is within range.
But At start time, There is problem if Open Manipulator's joint position is not within range.
If Open Manipulator's joint position is not within range at start time.
like below picture.  ( J1 : -0.078, J2 : -2.088, J3 : 1.531, J4 : 0.623 )
https://github.com/hyunoklee/pcsetting/blob/master/20181204_212515.jpg?raw=true
Then I can see the following Error log and anymore I can not control open manipulator by  GUI tool .
Init pose button and Home pose button also do not work and print log about checkLimit. 
/////////////ERRO LOG////////////////////////////////////////////////////////////////////////////////////
ROS_MASTER_URI=http://localhost:11311

process[open_manipulator_controller-1]: started with pid [2152]
Joint Dynamixel ID : 11, Model Name : XM430-W350
Joint Dynamixel ID : 12, Model Name : XM430-W350
Joint Dynamixel ID : 13, Model Name : XM430-W350
Joint Dynamixel ID : 14, Model Name : XM430-W350
Gripper Dynamixel ID : 15, Model Name :XM430-W350
[INFO] Successed to OpenManipulator initialization
[ERROR] [checkLimit] Goal value exceeded limit at joint2.
[ERROR] [checkLimit] Goal value exceeded limit at joint2.
[ERROR] [checkLimit] Goal value exceeded limit at joint2.

Affonso-Gui, 2018-12-05:
Had similar problems on https://github.com/ROBOTIS-GIT/open_manipulator/pull/46, which was solved by [this](https://github.com/ROBOTIS-GIT/open_manipulator/pull/46/files) commit.

It seems that the problem is that the joint limits defined in [open_manipulator_libs](https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_libs/src/OpenManipulator.cpp#L34-L82) are different from the corrected URDF ones.

gpwhd07, 2018-12-05:
Hi @hyunoklee 
Thank you for your feedback. We also think this is a very important issue.
We added a joint limit to prevent collisions between adjacent links.
As for the posture shown in the photograph, it can be interpreted that the collision between each component has already occurred [link](https://github.com/hyunoklee/pcsetting/blob/master/20181204_212515.jpg?raw=true).
We do not recommend starting the controls in this pose.
In this case, use the 'Actuator disable' button on the GUI to disable the torque of the dynamixel, position the OpenManipulator in a position where collision does not occur, and then re-enable it.

closed 2018-12-05 by hyunoklee