ROBOTIS-GIT/open_manipulator #68: Open_manipulator with AX-12 servos

sonary01, 2018-12-11:
I am using AX-12 motors for a customized open manipulator, which connected to PC with U2D2. I am facing problem while running the open_manipulator_controller.launch file.  Is it possible to use AX, MX servomotors with this package? Any help is much appreciated.
Below is the error message.

> started roslaunch server http://localhost:41105/
> 
> SUMMARY
> ========
> 
> PARAMETERS
>  * /open_manipulator_controller/baud_rate: 57142
>  * /open_manipulator_controller/usb_port: /dev/ttyUSB0
>  * /open_manipulator_controller/using_platform: True
>  * /robot_name: open_manipulator
>  * /rosdistro: kinetic
>  * /rosversion: 1.12.14
> 
> NODES
>   /
>     open_manipulator_controller (open_manipulator_controller/open_manipulator_controller)
> 
> ROS_MASTER_URI=http://localhost:11311
> 
> process[open_manipulator_controller-1]: started with pid [6286]
> [ERROR] [TxRxResult] There is no status packet!
> [ERROR] Please check your Dynamixel ID
> [ERROR] [TxRxResult] There is no status packet!
> [ERROR] Please check your Dynamixel ID
> [ERROR] [TxRxResult] There is no status packet!
> [ERROR] Please check your Dynamixel ID
> [ERROR] [TxRxResult] There is no status packet!
> [ERROR] Please check your Dynamixel ID
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [TxRxResult] There is no status packet!
> [ERROR] Please check your Dynamixel ID
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelDriver] Failed to get the Tool
> [ERROR] [DynamixelWorkbench] Failed to change torque status!
> [ERROR] [DynamixelWorkbench] Failed to change torque status!
> [ERROR] [DynamixelWorkbench] Failed to change torque status!
> [ERROR] [DynamixelWorkbench] Failed to change torque status!
> [ERROR] [DynamixelWorkbench] Failed to change torque status!
> [ERROR] [TxRxResult] There is no status packet!
> [ERROR] groupSyncRead getdata failed
> [ERROR] groupSyncRead getdata failed
> [ERROR] groupSyncRead getdata failed

routiful, 2018-12-11:
OpenManipulator supports Dynamixel with protocol 2.0.
AX can't work with protocol 2.0 but MX has different firmware work with it.
http://emanual.robotis.com/docs/en/dxl/mx/mx-64-2/

sonary01, 2018-12-12:
Hi, Thanks for the info. I tried by using motors MX-106, 64 and 28 motors with protocol 2.  IDs  are set from 1 to 5. The USB port is enabled. Still, no luck and getting the same error message. Someone else also asked a similar issue  #10, but no solutions.  Is there anything I can do to solve this problem?

routiful, 2018-12-13:
If you want to change Dynamixel ID, Please check [`OpenManipulator.cpp`](https://github.com/ROBOTIS-GIT/open_manipulator/blob/be2859a0506b4e941a19435c0a07562b41768a27/open_manipulator_libs/src/OpenManipulator.cpp#L40) in open_manipulator_lib folder. The default ID is that **joint is 11, 12, 13, 14 and gripper is 15**