ROBOTIS-GIT/open_manipulator #71: Error in XM430-W350 open serial motor.

Fred159, 2018-12-21:
ming@ming-Lenovo-ideapad-330-15ICH:~/opencr_update$ roslaunch open_manipulator_controller open_manipulator_controller.launch
... logging to /home/ming/.ros/log/dec163c4-04c5-11e9-ba1b-983b8fe14d12/roslaunch-ming-Lenovo-ideapad-330-15ICH-6049.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:40379/

SUMMARY
========

PARAMETERS
 * /joint_state_publisher/source_list: ['open_manipulato...
 * /joint_state_publisher/use_gui: False
 * /move_group/allow_trajectory_execution: False
 * /move_group/arm/default_planner_config: RRTConnect
 * /move_group/arm/longest_valid_segment_fraction: 0.005
 * /move_group/arm/planner_configs: ['SBL', 'EST', 'L...
 * /move_group/arm/projection_evaluator: joints(joint1,joi...
 * /move_group/capabilities: 
 * /move_group/disable_capabilities: 
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_safe_path_cost: 1
 * /move_group/planner_configs/BFMT/balanced: 0
 * /move_group/planner_configs/BFMT/cache_cc: 1
 * /move_group/planner_configs/BFMT/extended_fmt: 1
 * /move_group/planner_configs/BFMT/heuristics: 1
 * /move_group/planner_configs/BFMT/nearest_k: 1
 * /move_group/planner_configs/BFMT/num_samples: 1000
 * /move_group/planner_configs/BFMT/optimality: 1
 * /move_group/planner_configs/BFMT/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMT/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECE/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECE/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECE/range: 0.0
 * /move_group/planner_configs/BKPIECE/type: geometric::BKPIECE
 * /move_group/planner_configs/BiEST/range: 0.0
 * /move_group/planner_configs/BiEST/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRT/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRT/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRT/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRT/init_temperature: 100
 * /move_group/planner_configs/BiTRRT/range: 0.0
 * /move_group/planner_configs/BiTRRT/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRT/type: geometric::BiTRRT
 * /move_group/planner_configs/EST/goal_bias: 0.05
 * /move_group/planner_configs/EST/range: 0.0
 * /move_group/planner_configs/EST/type: geometric::EST
 * /move_group/planner_configs/FMT/cache_cc: 1
 * /move_group/planner_configs/FMT/extended_fmt: 1
 * /move_group/planner_configs/FMT/heuristics: 0
 * /move_group/planner_configs/FMT/nearest_k: 1
 * /move_group/planner_configs/FMT/num_samples: 1000
 * /move_group/planner_configs/FMT/radius_multiplier: 1.1
 * /move_group/planner_configs/FMT/type: geometric::FMT
 * /move_group/planner_configs/KPIECE/border_fraction: 0.9
 * /move_group/planner_configs/KPIECE/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECE/goal_bias: 0.05
 * /move_group/planner_configs/KPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECE/range: 0.0
 * /move_group/planner_configs/KPIECE/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECE/border_fraction: 0.9
 * /move_group/planner_configs/LBKPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/LBKPIECE/range: 0.0
 * /move_group/planner_configs/LBKPIECE/type: geometric::LBKPIECE
 * /move_group/planner_configs/LBTRRT/epsilon: 0.4
 * /move_group/planner_configs/LBTRRT/goal_bias: 0.05
 * /move_group/planner_configs/LBTRRT/range: 0.0
 * /move_group/planner_configs/LBTRRT/type: geometric::LBTRRT
[... 230 lines cut ...]
[ INFO] [1545359123.266184698]: Loading robot model 'open_manipulator'...
[ INFO] [1545359123.266217683]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ INFO] [1545359124.020115521]: Starting scene monitor
[ INFO] [1545359124.024352171]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1545359124.276935656]: Constructing new MoveGroup connection for group 'arm' in namespace ''
[ INFO] [1545359125.320060793]: Ready to take commands for planning group arm.
[ INFO] [1545359125.320151973]: Looking around: no
[ INFO] [1545359125.320219750]: Replanning: no
[ WARN] [1545359125.357431447]: Interactive marker 'EE:goal_end_effector_link' contains unnormalized quaternions. This warning will only be output once but may be true for others; enable DEBUG messages for ros.rviz.quaternions to see more details.

my linux is 16.04 LTS
ROS is kinetic 1.12.14 
using open cr 1.0

I can find dynamixel with below code with openCR firmware " usb_to_dxl"
rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyACM0

If I change the openCR firmware into "open_manipulator/example/arduino/chain" , the below code can't find dynamixel.
rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyACM0

What is the problem??

How to run below code with appropriate serial port?
roslaunch open_manipulator_controller open_manipulator_controller.launch
------------------------------

routiful, 2018-12-21:
Please check your Dynamixel ID and Baudrate.
The default ID is 11, 12, 13, 14, and 15, Baudrate is 1M

http://emanual.robotis.com/docs/en/platform/openmanipulator/#launch-controller

Fred159, 2018-12-21:
Sorry I am new to ROS.. So I still have problems to ask.
I'm not sure where should I change Dynamixel ID and Baudrate.
You mean I should change the dynamixel id with arduino OpenCR example/dynamixelworkbench/s_monitor? 
Then open serial monitor and change id and baud rate with serial command ? 
But the ID in serial monitor just shows Id: 1 ? how could I change 1 ID into 11, 12, 13, 14, and 15?

Fred159, 2018-12-21:
I wonder the only one ID I have is '1'.
So I change the only one dxl with ID '1' into '11' , and baud rate into 1000000.(in single controller gui).
Then how could I assign ID 12,13,14,15 ??

routiful, 2018-12-22:
Let's try step by step.

1. Disconnect all Dynamixels
2. Connect All Dynamixels to U2D2
3. Run find_dynamixel and please check ID and BaudRate

```
$ rosrun dynamixel_workbench_controllers find_dynamixel /dev/ttyACM0
```

6. If you find all Dynamixels, connect O.N.E Dynamixel and launch single_manager

```
$ roslaunch dynamixel_workbench_single_manager single_manager.launch /dev/ttyUSB0 57600
```

7. Change Dynamixel ID 11,12,13,14,15

Fred159, 2018-12-26:
Thank you. I will try this method first!

Fred159, 2018-12-31:
Thank you so much.
I solved the problem with your kind help.
All same as you suggest except step 6.
I change below command with port name of '/dev/ttyACM0`
`roslaunch dynamixel_workbench_single_manager single_manager.launch /dev/ttyUSB0 57600` 
However, errors happened. So I change my launch file directly(change device name to /dev/ttyACM0). And it works.
I'm going to learn more about dynamixelSDK. 
And after learn how to use dynamixelSDK in openmanipulator, I will try to control manipulator-H from robotis.
Thank you for your help again.