ROBOTIS-GIT/open_manipulator #73: Open Manipulator Controller with MoveIt option

fatihirim, 2018-12-31:
Hello

I try to launch Open Manipulator Controller with MoveIt Option. I follow the instructions on your e-manual page for Open Manipulator. However when I run he following command:

`roslaunch open_manipulator_controller open_manipulator_controller.launch use_moveit:=true`

I get the error:

```
started roslaunch server http://robut-linux-1:36685/

SUMMARY
========

PARAMETERS
 * /move_group/allow_trajectory_execution: False
 * /move_group/arm/default_planner_config: RRTConnect
 * /move_group/arm/longest_valid_segment_fraction: 0.005
 * /move_group/arm/planner_configs: ['SBL', 'EST', 'L...
 * /move_group/arm/projection_evaluator: joints(joint1,joi...
 * /move_group/capabilities: 
 * /move_group/disable_capabilities: 
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_safe_path_cost: 1
 * /move_group/planner_configs/BFMT/balanced: 0
 * /move_group/planner_configs/BFMT/cache_cc: 1
 * /move_group/planner_configs/BFMT/extended_fmt: 1
 * /move_group/planner_configs/BFMT/heuristics: 1
 * /move_group/planner_configs/BFMT/nearest_k: 1
 * /move_group/planner_configs/BFMT/num_samples: 1000
 * /move_group/planner_configs/BFMT/optimality: 1
 * /move_group/planner_configs/BFMT/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMT/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECE/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECE/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECE/range: 0.0
 * /move_group/planner_configs/BKPIECE/type: geometric::BKPIECE
 * /move_group/planner_configs/BiEST/range: 0.0
 * /move_group/planner_configs/BiEST/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRT/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRT/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRT/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRT/init_temperature: 100
 * /move_group/planner_configs/BiTRRT/range: 0.0
 * /move_group/planner_configs/BiTRRT/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRT/type: geometric::BiTRRT
 * /move_group/planner_configs/EST/goal_bias: 0.05
 * /move_group/planner_configs/EST/range: 0.0
 * /move_group/planner_configs/EST/type: geometric::EST
 * /move_group/planner_configs/FMT/cache_cc: 1
 * /move_group/planner_configs/FMT/extended_fmt: 1
 * /move_group/planner_configs/FMT/heuristics: 0
 * /move_group/planner_configs/FMT/nearest_k: 1
 * /move_group/planner_configs/FMT/num_samples: 1000
 * /move_group/planner_configs/FMT/radius_multiplier: 1.1
 * /move_group/planner_configs/FMT/type: geometric::FMT
 * /move_group/planner_configs/KPIECE/border_fraction: 0.9
 * /move_group/planner_configs/KPIECE/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECE/goal_bias: 0.05
 * /move_group/planner_configs/KPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECE/range: 0.0
 * /move_group/planner_configs/KPIECE/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECE/border_fraction: 0.9
 * /move_group/planner_configs/LBKPIECE/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/LBKPIECE/range: 0.0
 * /move_group/planner_configs/LBKPIECE/type: geometric::LBKPIECE
 * /move_group/planner_configs/LBTRRT/epsilon: 0.4
 * /move_group/planner_configs/LBTRRT/goal_bias: 0.05
 * /move_group/planner_configs/LBTRRT/range: 0.0
[... 105 lines cut ...]
ROS_MASTER_URI=http://localhost:11311

process[robot_state_publisher-1]: started with pid [31450]
process[move_group-2]: started with pid [31451]
process[rviz_robut_linux_1_31430_7019979553358795822-3]: started with pid [31453]
process[open_manipulator-4]: started with pid [31464]
[ INFO] [1546259564.320588893]: rviz version 1.12.16
[ INFO] [1546259564.320645622]: compiled against Qt version 5.5.1
[ INFO] [1546259564.320660222]: compiled against OGRE version 1.9.0 (Ghadamon)
Joint Dynamixel ID : 11, Model Name : XH430-W350
Joint Dynamixel ID : 12, Model Name : XM430-W350
Joint Dynamixel ID : 13, Model Name : XM430-W350
Joint Dynamixel ID : 14, Model Name : XH430-W350
[move_group-2] process has died [pid 31451, exit code -11, cmd /opt/ros/kinetic/lib/moveit_ros_move_group/move_group __name:=move_group __log:=/home/fatih/.ros/log/6cfd1c8a-0cf7-11e9-a895-7cb0c2fe1875/move_group-2.log].
log file: /home/fatih/.ros/log/6cfd1c8a-0cf7-11e9-a895-7cb0c2fe1875/move_group-2*.log
Gripper Dynamixel ID : 15, Model Name :XM430-W210

[ INFO] [1546259564.901577411]: Stereo is NOT SUPPORTED
[ INFO] [1546259564.901652515]: OpenGl version: 3 (GLSL 1.3).
[rviz_robut_linux_1_31430_7019979553358795822-3] process has died [pid 31453, exit code -11, cmd /opt/ros/kinetic/lib/rviz/rviz -d /home/fatih/WORKSPACES/ws_open_manipulator/src/open_manipulator/open_manipulator_moveit/launch/moveit.rviz __name:=rviz_robut_linux_1_31430_7019979553358795822 __log:=/home/fatih/.ros/log/6cfd1c8a-0cf7-11e9-a895-7cb0c2fe1875/rviz_robut_linux_1_31430_7019979553358795822-3.log].
log file: /home/fatih/.ros/log/6cfd1c8a-0cf7-11e9-a895-7cb0c2fe1875/rviz_robut_linux_1_31430_7019979553358795822-3*.log
```

Looks like ROS nodes are dying after started.
Can you help with this error?

fatihirim, 2018-12-31:
Also in the log of my ros-master, I have the following error:

```
[rosmaster.master][INFO] 2018-12-31 15:32:41,815: publisherUpdate[/tf_static] -> http://robut-linux-1:34029/ []
[rosmaster.master][INFO] 2018-12-31 15:32:41,815: publisherUpdate[/tf_static] -> http://robut-linux-1:34029/ []: sec=0.00, exception=[Errno 111] Connection refused
[rosmaster.threadpool][ERROR] 2018-12-31 15:32:41,815: Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosmaster/threadpool.py", line 218, in run
    result = cmd(*args)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rosmaster/master_api.py", line 210, in publisher_update_task
    ret = xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
    return self.__send(self.__name, args)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
    verbose=self.__verbose
  File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
    return self.single_request(host, handler, request_body, verbose)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1311, in single_request
    self.send_content(h, request_body)
  File "/usr/lib/python2.7/xmlrpclib.py", line 1459, in send_content
    connection.endheaders(request_body)
  File "/usr/lib/python2.7/httplib.py", line 1053, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 897, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 859, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 836, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 575, in create_connection
    raise err
error: [Errno 111] Connection refused
```

Hope this helps in identifying the root cause and a possible solution...

fatihirim, 2019-01-02:
I get the same error when trying to run demo under moveit package..

routiful, 2019-01-03:
```
NODES
  /
    move_group (moveit_ros_move_group/move_group)
    open_manipulator (open_manipulator_controller/open_manipulator_controller)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz_robut_linux_1_31430_7019979553358795822 (rviz/rviz)
```

joint_state_publisher was not launched as shown messages you attached.
When did you downloaded packages? check updates and open_manipulator_moveit.launch file

closed 2019-01-29 by gpwhd07