ROBOTIS-GIT/open_manipulator #76: Open Manipulator with TB3

Fred159, 2019-01-08:
Hello . 
I want to connect my remote pc with turtlebot  waffle with open manipulator. 
Each of part works well. 

But errors happened when I follow the manual. 
Error happened like below. 
So I checked , there is no <State.h> file, but file 'Pick.cpp' need 'state.h' file in package 'open_manipulator_with_tb3_tools'.
 
------
`
/home/ming/catkin_ws/src/open_manipulator_with_tb3/open_manipulator_with_tb3_tools/src/place.cpp:27:41: fatal error: open_manipulator_msgs/State.h: No such file or directory
compilation terminated.
/home/ming/catkin_ws/src/open_manipulator_with_tb3/open_manipulator_with_tb3_tools/src/pick.cpp:27:41: fatal error: open_manipulator_msgs/State.h: No such file or directory
compilation terminated.
open_manipulator_with_tb3/open_manipulator_with_tb3_tools/CMakeFiles/place.dir/build.make:62: recipe for target 'open_manipulator_with_tb3/open_manipulator_with_tb3_tools/CMakeFiles/place.dir/src/place.cpp.o' failed
make[2]: *** [open_manipulator_with_tb3/open_manipulator_with_tb3_tools/CMakeFiles/place.dir/src/place.cpp.o] Error 1
CMakeFiles/Makefile2:4718: recipe for target 'open_manipulator_with_tb3/open_manipulator_with_tb3_tools/CMakeFiles/place.dir/all' failed
make[1]: *** [open_manipulator_with_tb3/open_manipulator_with_tb3_tools/CMakeFiles/place.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
open_manipulator_with_tb3/open_manipulator_with_tb3_tools/CMakeFiles/pick.dir/build.make:62: recipe for target 'open_manipulator_with_tb3/open_manipulator_with_tb3_tools/CMakeFiles/pick.dir/src/pick.cpp.o' failed
make[2]: *** [open_manipulator_with_tb3/open_manipulator_with_tb3_tools/CMakeFiles/pick.dir/src/pick.cpp.o] Error 1
CMakeFiles/Makefile2:5023: recipe for target 'open_manipulator_with_tb3/open_manipulator_with_tb3_tools/CMakeFiles/pick.dir/all' failed
make[1]: *** [open_manipulator_with_tb3/open_manipulator_with_tb3_tools/CMakeFiles/pick.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j12 -l12" failed
`

routiful, 2019-01-08:
I have commented below.

https://github.com/ROBOTIS-GIT/open_manipulator_with_tb3/issues/6

routiful, 2019-01-14:
This issue was duplicated with https://github.com/ROBOTIS-GIT/open_manipulator_with_tb3/issues/6
If anyone want to follow it, please refer that issue page.

closed 2019-01-14 by routiful