ROBOTIS-GIT/open_manipulator #77: Consider Including URDF file in addition to XACRO

sea-bass, 2019-01-15:
Hello all,

I maintain another repository that uses the URDF/XACRO files in this one: https://www.mathworks.com/matlabcentral/fileexchange/65316-designing-robot-manipulator-algorithms

Right now, I have a script that takes the XACRO file and converts it to URDF with a few hard-coded changes, e.g., replacing expressions like "${PI}" with "3.1416"

Would it be possible to include an automatically generated URDF version of the XACRO file as well? You can do this with the "xacro" ROS package:

http://wiki.ros.org/xacro

Thanks, and happy to help with this as needed.

gpwhd07, 2019-01-29:
Hi @sea-bass 
Sorry for late reply.

First of all thank you for using our manipulator. 
We also fully agree with your opinion, but we do not have much time with other updates. So, if possible, I would appreciate it if you contributed to the above URDF/XACRO file updates.

Affonso-Gui, 2019-01-29:
Hello,

I have a few questions regarding this suggestion:
- Why would it be necessary to have the URDF file here, if we can automatically generate it (with, as you pointed, the xacro package) whenever it is needed? 
Having both the XACRO and URDF in the repository means that we have to maintain both files, which is error-prompt (e.g. the XACRO gets updated but forget to re-generate the URDF).
- Are there any limitations that prevent you from generating the URDF directly from your package/ getting it from the `robot_description` parameter?

sea-bass, 2019-01-29:
It's a valid point.

Basically, due to the MathWorks licensing, I am unable to directly share any of the source files in the submission, so I rely on automatically cloning this repo in my startup script and modifying the XACRO file to generate the URDF with a custom script.

The alternative is for the end user to do the XACRO to URDF conversion themselves, which is more work on them and additionally requires a ROS install, which most people don't have.

Agreed that it's just a convenience/enhancement. I have created a failsafe mechanism that loads a pre-saved 3D model if anything goes wrong in the process.

closed 2019-02-27 by gpwhd07