ROBOTIS-GIT/open_manipulator #81: About open_manipulator/option topic

Affonso-Gui, 2019-01-24:
If the `/open_manipulator/option` only answers to string data equal to "print_open_manipulator_setting", why don't we have it setted as an empty type topic or service?

Are there any plans to add other accepted commands to this topic?

gpwhd07, 2019-01-29:
Hi @Affonso-Gui 

Yes you are right.
We have a plan to add another option using `/open_manipulator/option`, so we set the topic to string data.

Affonso-Gui, 2019-01-29:
I see.

It would also be nice to have some warning telling the user which commands are supported (something like the old operating mode: https://github.com/ROBOTIS-GIT/open_manipulator/blob/49fce7592f75b88de64b7cbca8dfab539b4e701f/open_manipulator_dynamixel_ctrl/src/dynamixel_controller.cpp#L120-L136)

gpwhd07, 2019-01-29:
Hi, again,

Currently we do not use the dynamixel_controller package at all. This package is not currently running because it was not updated together when the dynamixel_workbench package was updated.

Affonso-Gui, 2019-01-29:
Sorry for the misleading comment.

I just intended to show an example of case handling that notices the user about which commands are available if no match is found.

In other words, it would be nice to add an else clause on the below (I can make a PR for that too, if that is the case)
https://github.com/ROBOTIS-GIT/open_manipulator/blob/3b575b4c6b4c0212713c1e71a74da23e9192c227/open_manipulator_controller/src/open_manipulator_controller.cpp#L197-L201

gpwhd07, 2019-02-01:
Thank you for your proposition.
We updated the OpenManipulator packages yesterday. 
We changed  `OM_CONTROLLER::printManipulatorSettingCallback` to `OpenManipulatorController::openManipulatorOptionCallback`.

```
void openManipulatorController::openManipulatorOptionCallback(const std_msgs::String::ConstPtr &msg)
{
  if(msg->data == "print_open_manipulator_setting")
    open_manipulator_.printManipulatorSetting();
}
```

I think this update helped users to understand that this message could be used to configure other options, but adding `else` clause would be a good way to help users understand.
If you think you need to add only `else` clause to help users understanding, we are willing to add 'else' without some function. If not, we will add it when other additional options are available.

Affonso-Gui, 2019-02-07:
I believe that adding an `else` clause would be clarifying (even with only one option).

However, this is not a priority issue, so if we are having updates soon it should be ok to wait for them.

gpwhd07, 2019-02-12:
I understand your suggestion, and agree.
We released first version(1.0.0) of this package, we didn't added `else` clause at this version.
But we will added new option soon, it will be included next version.

Thanks.
I will close this issue once, but you can reopen if there are additional requests.

closed 2019-02-12 by gpwhd07