ROBOTIS-GIT/open_manipulator #88: Joints position are not in sync with simulations

LJM2018, 2019-02-07:
Hello everyone, I was able to control each joint of the open manipulator after I connected the gripper motor to the chain. However, in the simulations(Processing, Rviz), the joints positions are definitely not synced. There is a difference in joint 3(the third actuator counted from the base). Is it a software issue or is it a issue that can be solved by doing calibration with R+ manager? 

[image]

below shows the offset between the actual pose and the wrong calculated pose.

[image]

Also, since I couldn't find my calibration adapter, I wonder if there is a STL file that allows me to 3D print it. Thanks in advance.

Affonso-Gui, 2019-02-08:
Hello,

Are the above images taken in the same timing? (that is, do the Rviz and Processing show the same offset for joint 3, or each one has a different error value?)

Besides the offset, is the Rviz responsive? When you turn off the motors and move the manipulator around is the Rviz model correctly updated? Specially for joint 3, moving it upwards make the joint 3 in the model move in the same direction/ somewhat the same value?

If all the answers are 'yes', then it seems to be an offset problem. Make sure that the zero point of the joint 3 actuator (indicated by a dot in the servo horn) is pointing straight to the next link and that the [homing offset](http://support.robotis.com/en/product/actuator/dynamixel_x/xm_series/xm430-w350.htm#bookmark7) is set to zero.

LJM2018, 2019-02-08:
> Hello,
>
> Are the above images taken in the same timing? (that is, do the Rviz and Processing show the same offset for joint 3, or each one has a different error value?)
>
> Besides the offset, is the [...]

Hello  @Affonso-Gui, thanks for your reply!. 

Both Processing and Rviz show the same but wrong configuration of the manipulator. They are all responsive to change of the joint angle.

[image]

However,
in Processing IDE, the change value of joint positions updated correctly but the direction is wrong(opposite) for most of the actuators(joint 1 seems correct).
In Rviz, the situation is similar, even though the simulation updated the change of value correctly, the direction of changing joint angle is opposite.

[image]

I wonder if this is due to some installation mistakes because this manipulator is not installed by myself. Do you have any suggestions? Thank you!

gpwhd07, 2019-02-12:
Hello @LJM2018 
and thanks @Affonso-Gui,

I think this problem is due to installation mistakes such like of installed direction of Dynamixel is wrong.
Please check manipulator installation about specially `dynamixel installation direction` and `Horn Align`, refer to [assembly manual of e-manual](http://emanual.robotis.com/docs/en/platform/openmanipulator/#assembly-manual).

LJM2018, 2019-02-12:
Thank you so much! @gpwhd07 I just re-assembled it and everything works fine.

closed 2019-02-12 by LJM2018