ROBOTIS-GIT/open_manipulator #89: How to control turtlebot's wheels and manipulator at the same time?

LJM2018, 2019-02-11:
I was trying to teleop turtlebot3's wheels in order to let it move using "roslaunch turtlebot3_teleop keyboard.launch" after running "roslaunch open_manipulator_with_tb3_tools robot.launch". The OpenCR board is uploaded with open_manipulator_with_tb3_core. However, even though I was able to control the manipulator by doing rosserivce call ....   I wasn't able to let the turtlebot's wheels moving using the above command. Since that command is not actually designed for turtlebot with manipulator, I wonder if there is a better way to control manipulator and turtlebot's wheels (via keyboard..etc).

LJM2018, 2019-02-11:
I couldn't find any other information about this on the Robotis website.

routiful, 2019-02-12:
You can find some tutorials to follow emanual
http://emanual.robotis.com/docs/en/platform/turtlebot3/manipulation/#manipulation

LJM2018, 2019-02-13:
Hello, thanks for the reply. I have read the entire emanual, however, I still couldn't figure out how to control joints of the manipulator when the wheels are controlled at the same time. 

Is there a way to control the manipulator(not just gripper) with "open_manipulator_with_tb3_core" uploaded to turtlebot using **just the terminal**? Because what I saw in the tutorial is only about how to use moveit and its interactive ball to control the manipulator.
[image]

(I tired rosservice call /om_with_tb3/.. to control those joints like controlling the gripper but failed)
[image]

The official tutorial only shows how to control the gripper using Moveit and some task controller but there is almost no information about controlling the other joints. I tried 'rostopic list' as well as 'rosservice list' but I only found some related topics/services about the gripper's **position**, there are nothing related to joint's position.

LJM2018, 2019-02-13:
Is there a XXX_teleop.launch that can be used to control both wheels and manipulator at the same time?

routiful, 2019-02-14:
1. OpenCR core has subscriber to control manipulator and gripper. The releated topic names are `/joint_trajectory_point` and `/gripper_position`.

2. When you launch manipulation.launch, `moveit_bridge` node will run as well. This node which includes moveit api will makes planned path to get desired goal. the releated service names are `/moveit/get_joint_position` and `/moveit/get_kinematics_pose`

3. teleop.launch publishes `/cmd_vel` msg only.

LJM2018, 2019-02-16:
When I launch the moveit (roslaunch open_manipulator_with_tb3_tools manipulation.launch use_platform:=true), I got errors like below and there was no interactive marker on the moveit interface

[image]

[image]

I feel like the moveit isn't completely launched since there is no way to control the manipulator. Do you have any suggestions?

Any suggestion will be appreciated! :)

LJM2018, 2019-02-16:
> 1. OpenCR core has subscriber to control manipulator and gripper. The releated topic names are `/joint_trajectory_point` and `/gripper_position`.
> 2. When you launch manipulation.launch, [...]

Thank you for your reply!

closed 2019-02-27 by gpwhd07