ROBOTIS-GIT/open_manipulator #91: OpenCR firmware for OpenManipulator with ROS

LucidOne, 2019-02-25:
I am unable to find documentation for [open_manipulator_chain_ROS](https://github.com/ROBOTIS-GIT/OpenCR/tree/master/arduino/opencr_arduino/opencr/libraries/OpenManipulator/example/Chain/open_manipulator_chain_ROS).
Is this a project under development to replace [usb_to_dxl](https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/OpenCR/examples/10.%20Etc/usb_to_dxl/usb_to_dxl.ino)?

If so, when might documentation become available?

gpwhd07, 2019-02-25:
Hi @LucidOne 

We will support both of `usb_to_dxl` and  `open_manipulator_chain_ROS`, and update the documentation of the `open_manipulator_chain_ROS` soon.

If you want to use `open_manipulator_chain_ROS` before the documentation updates, I can tell you.

gpwhd07, 2019-03-14:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-03-14 by gpwhd07