ROBOTIS-GIT/open_manipulator #97: Can't find open_manipulator_demo.launch

loseeveryday, 2019-03-28:
I am following the ROS_Robot_Programming_CN and learning to control the manipulator in gazebo from rviz. But I find that I can't find files such as open_manipulator_demo.launch under the folder open_manipulator_moveit , position_controller.cpp and so on.Have the code changed?But I can't find files similar with them. Can anyone give me some suggestion? Thanks a lot.

gpwhd07, 2019-03-28:
Hi, @loseeveryday 

> Have the code changed?
Yes, we updated OpenManipulator controller packages. 
You can refer our [e-manual](http://emanual.robotis.com/docs/en/platform/openmanipulator_x/overview/) for OpenManipulator.

In conclusion, you can use the following command to open same launch file with open_manipulator_demo.launch
`roslaunch open_manipulator_controller open_manipulator_controller.launch use_moveit:=true`

If you want to use it with gazebo, use the following command.
`roslaunch open_manipulator_controller open_manipulator_controller.launch use_moveit:=true use_platform:=false`

loseeveryday, 2019-03-28:
Thanks for your reply! I will try it latter.
I have found those folder in the older version.

closed 2019-03-28 by loseeveryday