ROBOTIS-GIT/open_manipulator #98: Payload capacity of Open Manipulator

MONICA-PALLA, 2019-04-02:
May I know the payload which an Open Manipulator made of XL430 servos is capable of carrying?

Thank you,
Monica, P.

gpwhd07, 2019-04-04:
Hi, @MONICA-PALLA 

We do not officially support the OpenManipulator with XL430 because of the low payload.
We assume that its payload will be less than 50g.

Thanks

MONICA-PALLA, 2019-04-04:
Thank you for your reply.

Regards,
Monica, P.

closed 2019-04-04 by MONICA-PALLA

zlatinovStefan, 2019-10-07:
What about a combination of XM430 and XL430. Is it okay to replace XM430 with XL430 for the gripper or the waist?

gpwhd07, 2019-10-08:
Hi, @zlatinovStefan 

We haven't measured the payload with that system, but that might be possible. I think you can replace `first` and `fourth` XM430 to XL430. 
The gripper dynamixel have to use current based position control mode but XL430s are not provided that mode, so that you need to use XM430 for the gripper dynamixel.