ROBOTIS-GIT/open_manipulator #99: xl430

unverciftci, 2019-04-06:
I am using 5 xl430 for openmanipulator and I want to control it with opencr and processing. it happens that only the gripper works with the gui but the other joints do not respond.

gpwhd07, 2019-04-08:
Hi, @unverciftci 

Thanks for using our products.
For now, we don't support the OpenManipulator model using xl430.
But, you will be able to use it for simple motions playing. 

> [ERROR] [DynamixelWorkbench] Failed to set Current Based Position Control Mode!
This error occurs, because the dynamixel xl-series can't support `current based position control mode` used for gripper. But it will not affect the simple operation of the gripper dxl.

>The other joints do not respond
It is difficult to determine the cause of this problem just by the information now.
Please make sure you set the dynamixel's id and bard rate properly. 
If the settings are correct, please explain the details.

unverciftci, 2019-04-09:
Please remember your promise:
https://www.facebook.com/robotis.company/videos/openmanipulator-official-kit-version-assembly-video/552008381897347/

gpwhd07, 2019-04-10:
OpenManipulator made by xl430 does not support `current based position control mode` but it will works normally about the other functions. 

Even with the Xl series, all joint dynamixels of the OpenManipulator will function normally.
Please check again you set the dynamixel's id(11~15) and bard rate(1M) properly, and try to recover dynamixels. If the settings are correct, please tell me about the details.

closed 2019-04-22 by gpwhd07