ROBOTIS-GIT/open_manipulator #100: Errors that occur while running Gazebo : No P gain ~

JoDongWoon, 2019-04-22:
Hello~ I installed the following packages in the Ubuntu 18.04 environment and then launched the Gazebo
$ sudo apt-get install ros-melodic-ros-controllers ros-melodic-gazebo* ros-melodic-moveit*
$ cd ~/catkin_ws/src
$ git clone -b kinetic https://github.com/ros-industrial/industrial_core.git
$ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/DynamixelSDK.git
$ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/dynamixel-workbench.git
$ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/dynamixel-workbench-msgs.git
$ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/open_manipulator.git
$ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/open_manipulator_msgs.git
$ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/open_manipulator_simulations.git
$ git clone -b melodic-devel https://github.com/ROBOTIS-GIT/robotis_manipulator.git
$ sb & cm
$ roslaunch open_manipulator_gazebo open_manipulator_gazebo.launch

SUMMARY
========

PARAMETERS
 * /gazebo/enable_ros_network: True
 * /gazebo_ros_control/pid_gains/gripper/antiwindup: False
 * /gazebo_ros_control/pid_gains/gripper/d: 500.0
 * /gazebo_ros_control/pid_gains/gripper/i: 1000.0
 * /gazebo_ros_control/pid_gains/gripper/i_clamp: 0.2
 * /gazebo_ros_control/pid_gains/gripper/p: 100000.0
 * /gazebo_ros_control/pid_gains/gripper/publish_state: True
 * /gazebo_ros_control/pid_gains/gripper_sub/antiwindup: False
 * /gazebo_ros_control/pid_gains/gripper_sub/d: 1000.0
 * /gazebo_ros_control/pid_gains/gripper_sub/i: 250.0
 * /gazebo_ros_control/pid_gains/gripper_sub/i_clamp: 0.2
 * /gazebo_ros_control/pid_gains/gripper_sub/p: 100000.0
 * /gazebo_ros_control/pid_gains/gripper_sub/publish_state: True
 * /gazebo_ros_control/pid_gains/joint1/antiwindup: False
 * /gazebo_ros_control/pid_gains/joint1/d: 100.0
 * /gazebo_ros_control/pid_gains/joint1/i: 1000.0
 * /gazebo_ros_control/pid_gains/joint1/i_clamp: 0.2
 * /gazebo_ros_control/pid_gains/joint1/p: 100000.0
 * /gazebo_ros_control/pid_gains/joint1/publish_state: True
 * /gazebo_ros_control/pid_gains/joint2/antiwindup: False
 * /gazebo_ros_control/pid_gains/joint2/d: 500.0
 * /gazebo_ros_control/pid_gains/joint2/i: 1000.0
 * /gazebo_ros_control/pid_gains/joint2/i_clamp: 0.2
 * /gazebo_ros_control/pid_gains/joint2/p: 100000.0
 * /gazebo_ros_control/pid_gains/joint2/publish_state: True
 * /gazebo_ros_control/pid_gains/joint3/antiwindup: False
 * /gazebo_ros_control/pid_gains/joint3/d: 500.0
 * /gazebo_ros_control/pid_gains/joint3/i: 1000.0
 * /gazebo_ros_control/pid_gains/joint3/i_clamp: 0.2
 * /gazebo_ros_control/pid_gains/joint3/p: 100000.0
 * /gazebo_ros_control/pid_gains/joint3/publish_state: True
 * /gazebo_ros_control/pid_gains/joint4/antiwindup: False
 * /gazebo_ros_control/pid_gains/joint4/d: 500.0
 * /gazebo_ros_control/pid_gains/joint4/i: 1000.0
 * /gazebo_ros_control/pid_gains/joint4/i_clamp: 0.2
 * /gazebo_ros_control/pid_gains/joint4/p: 100000.0
 * /gazebo_ros_control/pid_gains/joint4/publish_state: True
 * /open_manipulator/gripper_position/joint: gripper
 * /open_manipulator/gripper_position/type: position_controll...
 * /open_manipulator/gripper_sub_position/joint: gripper_sub
 * /open_manipulator/gripper_sub_position/type: position_controll...
 * /open_manipulator/joint1_position/joint: joint1
 * /open_manipulator/joint1_position/type: position_controll...
 * /open_manipulator/joint2_position/joint: joint2
 * /open_manipulator/joint2_position/type: position_controll...
 * /open_manipulator/joint3_position/joint: joint3
 * /open_manipulator/joint3_position/type: position_controll...
 * /open_manipulator/joint4_position/joint: joint4
 * /open_manipulator/joint4_position/type: position_controll...
 * /open_manipulator/joint_state_controller/publish_rate: 1000
 * /open_manipulator/joint_state_controller/type: joint_state_contr...
 * /robot_description: <?xml version="1....
 * /rosdistro: melodic
 * /rosversion: 1.14.3
 * /use_sim_time: True

NODES
  /
    gazebo (gazebo_ros/gzserver)
    gazebo_gui (gazebo_ros/gzclient)
    urdf_spawner (gazebo_ros/spawn_model)
  /open_manipulator/
    controller_spawner (controller_manager/spawner)
    gripper_sub_publisher (open_manipulator_gazebo/gripper_sub_publisher)

auto-starting new master
process[master]: started with pid [7651]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to aa6a9d54-64d6-11e9-84ad-e82a448a8dc5
process[rosout-1]: started with pid [7662]
started core service [/rosout]
process[gazebo-2]: started with pid [7669]
process[gazebo_gui-3]: started with pid [7674]
process[urdf_spawner-4]: started with pid [7679]
process[open_manipulator/controller_spawner-5]: started with pid [7680]
process[open_manipulator/gripper_sub_publisher-6]: started with pid [7681]
[INFO] [1555920877.705735, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1555920877.985260193]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1555920877.986951011]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1555920878.060406160]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1555920878.062113348]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[INFO] [1555920878.297025, 0.000000]: Loading model XML from ros parameter robot_description
[INFO] [1555920878.313919, 0.000000]: Waiting for service /gazebo/spawn_urdf_model
[INFO] [1555920878.618052, 0.000000]: Calling service /gazebo/spawn_urdf_model
[INFO] [1555920878.918211, 0.000000]: Spawn status: SpawnModel: Successfully spawned entity
[ INFO] [1555920879.083849301]: Loading gazebo_ros_control plugin
[ INFO] [1555920879.084073021]: Starting gazebo_ros_control plugin in namespace: open_manipulator
[ INFO] [1555920879.084658127]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[urdf_spawner-4] process has finished cleanly
log file: /home/whehddns2/.ros/log/aa6a9d54-64d6-11e9-84ad-e82a448a8dc5/urdf_spawner-4*.log
**[ERROR] [1555920879.206903100]: No p gain specified for pid.  Namespace: /open_manipulator/gazebo_ros_control/pid_gains/joint1
[ERROR] [1555920879.208206473]: No p gain specified for pid.  Namespace: /open_manipulator/gazebo_ros_control/pid_gains/joint2
[ERROR] [1555920879.209289278]: No p gain specified for pid.  Namespace: /open_manipulator/gazebo_ros_control/pid_gains/joint3
[ERROR] [1555920879.210309576]: No p gain specified for pid.  Namespace: /open_manipulator/gazebo_ros_control/pid_gains/joint4
[ERROR] [1555920879.211646010]: No p gain specified for pid.  Namespace: /open_manipulator/gazebo_ros_control/pid_gains/gripper
[ERROR] [1555920879.212790568]: No p gain specified for pid.  Namespace: /open_manipulator/gazebo_ros_control/pid_gains/gripper_sub**
[ INFO] [1555920879.220204426]: Loaded gazebo_ros_control.
[INFO] [1555920879.522182, 0.000000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1555920879.532287, 0.000000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1555920879.546549, 0.000000]: Loading controller: joint_state_controller
[INFO] [1555920879.589858, 0.000000]: Loading controller: joint1_position
[INFO] [1555920879.645711, 0.000000]: Loading controller: joint2_position
[INFO] [1555920879.678206, 0.000000]: Loading controller: joint3_position
[INFO] [1555920879.727092, 0.000000]: Loading controller: joint4_position
[INFO] [1555920879.764402, 0.000000]: Loading controller: gripper_position
[INFO] [1555920879.797502, 0.000000]: Loading controller: gripper_sub_position
[INFO] [1555920879.835591, 0.000000]: Controller Spawner: Loaded controllers: joint_state_controller, joint1_position, joint2_position, joint3_position, joint4_position, gripper_position, gripper_sub_position

There's no problem running Gazebo yet, Can you tell me which file is relevant and how to fix it?

gpwhd07, 2019-04-30:
Hi @JoDongWoon 
Sorry to late reply.
I checked about this error, it is cause by it can not read PID gains in position controller of updated gazebo. You can fix that as modifying the `open_manipulator_controller.yaml` file of open_manipulator_gazebo pkg and  `open_manipulator.urdf.xacro` of open_manipulator_description pkg as following

`open_manipulator_description/urdf/open_manipulator.urdf.xacro`
```
<?xml version="1.0"?>
<!-- Open_Manipulator Chain -->
<robot name="open_manipulator" xmlns:xacro="http://ros.org/wiki/xacro">

  <!-- Import all Gazebo-customization elements, including Gazebo colors -->
  <xacro:include filename="$(find open_manipulator_description)/urdf/open_manipulator.gazebo.xacro" />
  <!-- Import Rviz colors -->
  <xacro:include filename="$(find open_manipulator_description)/urdf/materials.xacro" />

  <!-- Transmission macro -->
  <xacro:macro name="SimpleTransmission" params="joint n">
    <transmission name="tran${n}">
      <type>transmission_interface/SimpleTransmission</type>
      <joint name="${joint}">
        <hardwareInterface>hardware_interface**/EffortJointInterface**</hardwareInterface>
      </joint>
      <actuator name="motor${n}">
        <hardwareInterface>hardware_interface**/EffortJointInterface**</hardwareInterface>
        <mechanicalReduction>1</mechanicalReduction>
      </actuator>
    </transmission>
  </xacro:macro>

  <!-- World -->
  <link name="world">
  </link>

  <!-- World fixed joint-->
  <joint name="world_fixed" type="fixed">
    <origin xyz="0 0 0" rpy="0 0 0"/>
    <parent link="world"/>
    <child link="link1"/>
  </joint>

  <!-- Link 1 -->
  <link name="link1">
    <visual>
      <origin xyz="0 0 0" rpy="0 0 0"/>
      <geometry>
        <mesh filename="package://open_manipulator_description/meshes/chain_link1.stl" scale="0.001 0.001 0.001"/>
      </geometry>
      <material name="grey"/>
    </visual>

    <collision>
      <origin xyz="0 0 0" rpy="0 0 0"/>
      <geometry>
        <mesh filename="package://open_manipulator_description/meshes/chain_link1.stl" scale="0.001 0.001 0.001"/>
      </geometry>
    </collision>

<!--    <inertial>
      <origin xyz="0 0 0" />
      <mass value="0.082" />
      <inertia ixx="0.1" ixy="0.0" ixz="0.0"
               iyy="0.1" iyz="0.0"
               izz="0.1" />
    </inertial>-->

     <inertial>
      <origin xyz="3.0876154e-04 0.0000000e+00 -1.2176461e-04" />
      <mass value="7.9119962e-02" />
      <inertia ixx="1.2505234e-05" ixy="0.0" ixz="-1.7855208e-07"
               iyy="2.1898364e-05" iyz="0.0"
[... 316 lines cut ...]
# Position Controllers ---------------------------------------
joint1_position:
  type: **effort_controllers**/JointPositionController
  joint: joint1

joint2_position:
  type: **effort_controllers**/JointPositionController
  joint: joint2

joint3_position:
  type: **effort_controllers**/JointPositionController
  joint: joint3

joint4_position:
  type: **effort_controllers**/JointPositionController
  joint: joint4

gripper_position:
  type: **effort_controllers**/JointPositionController
  joint: gripper

gripper_sub_position:
  type: **effort_controllers**/JointPositionController
  joint: gripper_sub
```

gpwhd07, 2019-05-08:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-05-08 by gpwhd07

malihemarzani, 2023-09-18:
hello, while launching open_manipulator_gazebo.launch i will get the same error 
[ERROR] [1695051052.353447258]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint1
[ERROR] [1695051052.353732796]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint2
[ERROR] [1695051052.354036640]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint3
[ERROR] [1695051052.354431376]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/joint4

could you please confirm the solution, please?