ROBOTIS-GIT/open_manipulator #101: Overload

LucidOne, 2019-05-13:
In testing the Open Manipulator arm we have had Dynamixel ID 12 & 13 (shoulder & elbow) experience Overload (OL) errors when jogging the arm downward with keyboard teleop.

It looks like the arm should have some acceleration limiting to prevent this,.

gpwhd07, 2019-05-15:
Hi @LucidOne
Thank for using and test our OpenMANIPULATOR-X.

I want to check about your environment.
Did you used XM-series dynamixel for open manipulator?
If you used XL-series, such problems may occur due to lack of torque of the dynamixel.

If you want to change acceleration, you can test again after changing 'PATH_TIME' parameter in open_manipulator_teleop_joystick.h of teleop package to over than 1.0? We will also test about that.

LucidOne, 2019-05-15:
The arm used XL-series Dynamixels, I think they are a little weak for the shoulder and elbow.
Personally I'd prefer if the OpenMANIPULATOR-X shipped with 2 XM-series and 3 XL-series.

We are in the middle of other work at the moment but if you leave this open we can test 'PATH_TIME' to see how it performs.

gpwhd07, 2019-05-15:
Thank you for quick response.
We are also aware of the XL version of the lack of torque and are looking for measures.
For this reason, we are officially supporting only the XM version, not the XL version. 
Please change that two Dynamixels to XM, or adjust the path time.

gpwhd07, 2019-05-27:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-05-27 by gpwhd07