ROBOTIS-GIT/open_manipulator #102: F020 Errors

ed8metzler, 2019-05-14:
1. I have an OpenManipulator kit fully assembled. Dynamixel s are configured as 11-15 and are set to 1000000 baud.  I have Openmanipulator fully functional with U2D2. That is, GUI works, Viz, Keyboard and MoveIt all come up without errors. And Openmanipulator moves.
The problem I have is intermittent F020 errors with OpenCR losing connection to ACM0. Or complaining about sketch transfer errors. Arduino 1.8.9 installed. Simple “blink” test will work one time then fail next time. None of the Openmanipulator sketches work but sometimes they intermittently end with”jump to firmware” message (success). But next time fail to load/run with F020 error. in any case, nothing happens with OpenManipulator on OpenCR. 

2. Running Ubuntu 16.04 and ROS Kinetic. Have not tried with Windows. I have three different computers built up this way. One is running Ubuntu in a VM and the other two computers running Ubuntu as the native OS.

3. Have repeatedly completed Firmware Recovery Mode as shown in OpenManipulator X instructions.

Is it possible that many of the Examples on Arduino are programmed with the incorrect baud rate or that there is serious timing problem syncing commands between PC and OpenCR?

gpwhd07, 2019-05-15:
Hi @ed8metzler 
Thanks for using our openMANIPULATOR.

Did you upload the open_manipulator_chain example source code?
'OpenCR' source code we provide is  operatied without ROS packages, in the other word, the controller is working on OpenCR board, the control commends are sended through the usb cable by using `Processing IDE` program. 

If you are operating open_manipulator above environments, first of all, please check usb cable is stable, let me know which source code was uploaded to OpenCR.

ed8metzler, 2019-05-15:
Hi@Hye-Jong KIM,

 

Here are the steps I followed:
Select CHAIN from Examples on Arduino menu.
Compile and Load – sometimes I receive Jump to FW message; sometimes xF020 error.
OpenManipulator does nothing. 
Have replaced USB cable. No difference. 
To answer the question about Source Code – it is the same that is in the Examples (OpenCR is selected as Board under Tools menu). And selected the .JSON web site shown at the OpenManipulator X instructions.
Sometimes “Blink” and “Serial monitoring of SW1 setting” work. But most of the time not.
There is no consistency with OpenCR. One time I receive xF020 error; one time it gets jump to FW message.
 

My version of OpenCR seems to be 00. Are there other versions?

 

Ed Metzler

OpusK, 2019-05-15:
Hi, @ed8metzler 

This phenomenon is a phenomenon that is often covered in OpenCR issues.
0xF020 Error is a timeout error.
This is caused by uploading firmware through usb, which can be caused by various reasons such as OS, USB cable, equipment and so on.

Would you like to attach all the error messages (error log of the Arduino IDE) that came with the 0xF020 error?

ed8metzler, 2019-05-27:
Dear Kei:

 

I finally got OpenCR to work with OpenManipulator. It typically requires regular re-loading of firmware and doing the Recovery mode using buttons on the board. And pulling and putting back in USB cable. These have to be done in the appropriate sequence. And then it requires multiple retries to get jump_to_FW message. Afterwards, running the Processing program seems to be more reliable. But sometimes will not work depending on initial position of OpenManipulator. 

 

There are obviously synchronization issues between OpenCR, PC and OpenManipulator. Why has Robotis not fixed these issues if they are well known? 

 

Ed Metzler

OpusK, 2019-05-28:
Hi, @ed8metzler 

Uploading problems between OpenCR and PC may be caused by problems such as USB cable as well as OS problems.
This is a problem that can not be reproduced even with known contents because it differs from environment to environment. (Even though this phenomenon is repeated in your environment, it may not happen at all on the developer's PC)
There have been some people who have experienced this phenomenon so far, but we have never succeeded in reproducing it.

Anyway, there are difficulties in that we can not debug the same phenomenon.

> Afterwards, running the Processing program seems to be more reliable. But sometimes will not work depending on initial position of OpenManipulator.

In relation to this, @gpwhd07, please check this out.

gpwhd07, 2019-05-28:
Hi @ed8metzler 

> Afterwards, running the Processing program seems to be more reliable. But sometimes will not work depending on initial position of OpenManipulator.

I think this problem is due to some joints exceeding the limit of the joints in the initial position you set. I recommend to place OpenMANIPULATOR-X at the following pose where each component does not conflict, so that start the controller (Push the reset button of OpenCR).
[image]

gpwhd07, 2019-08-16:
This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-08-16 by gpwhd07

ed8metzler, 2019-08-16:
Your suggestion worked. Sorry I did not let you know.

Best wishes,

Ed Metzler