ROBOTIS-GIT/open_manipulator #103: I have a some question

hyunoklee, 2019-05-14:
hello, I have a some question. 
Recently I controled open manipulat by below command. 

roslaunch open_manipulator_controller open_manipulator_controller.launch
roslaunch open_manipulator_control_gui open_manipulator_control_gui.launch

A few weeks ago , I know that default value of use_moveit parameter  is false in open_manipulator_controller.launch file .  
So, I think open manipulator don't use moveit package in default command. 
But When I send inverse kinemetrics coordination to open manipulator by GUI tool, the open manipulator move to exact position. 

How does open manipulator work with inverse kinemetics without using MoveIt packages ? I would appreciate it if you let me know.

gpwhd07, 2019-05-15:
Hi @hyunoklee 

Yes, open_manipulator_controller works without Moveit package in default setting.
The inverse and forward kinematics source code is in open_manipulator_libs package, open manipulator is operated by using this source when move to a target position in cartesian coordinates.
We provide several kinematics solver algorisms, and you can check this source.
You can design new solver class, and apply it to open manipulator.
Please refer e-manual of robotis_manipulator about how to make and modify open_manipulator_libs.

http://emanual.robotis.com/docs/en/software/robotis_manipulator_libs/

JoDongWoon, 2019-05-15:
Can you tell me why you are not running MoveIt? Is not MoveIt unable to solve all the problems?

hyunoklee, 2019-05-15:
Thank you very much for your answer. :) 
Could you tell me why you did not use MoveIt as a default ?

gpwhd07, 2019-05-15:
Hi @JoDongWoon and @hyunoklee 

You are right, Moveit! can provide kinematics solving, path generation and so on, and its path generation performance is outstanding.
So we also provide methods to use moveit package.

But, if our manipulator only support Moveit! package, we can not implement a manipulator controller package that we are targeting.

- We want the framework that user can easily make and apply their own algorism such as kinematics solver and trajectory generator and so on for their research. MoveIt! also support to change several kinematics solver and path generator, but it is not simple that user make the solvers and generators.

- We want that open manipulator can be operated by same source codes on ROS and embedded board without ROS. The user need ROS environments to use Moveit! and that is not suitable to embedded board.

For these reasons and so on, we have created our own proprietary library that does not use moveit!

hyunoklee, 2019-05-15:
thank you very much :)

closed 2019-05-15 by hyunoklee

JoDongWoon, 2019-05-15:
I have come to know new facts. thank you for telling me