ROBOTIS-GIT/open_manipulator #104: Misleading open_manipulator_moving_state

Affonso-Gui, 2019-05-16:
`open_manipulator_moving_state` reported on `open_manipulator/states` seems to be based solely on the controller commands, rather than the dynamixel's **`Moving`** value (control table 122).

Because of that, the topic erroneously inform that the manipulator is moving when it is actually stopped  with servo off; or that it is stopped when it is being moved by a person.

gpwhd07, 2019-05-16:
Hi @Affonso-Gui :-)

Yes you are right about `open_manipulator_moving_state` reported on `open_manipulator/states`.
The `open_manipulator_moving_state` is possible to use the dynamixel's Moving value (control table 122) for that, and that method may be more suitable to OpenMANIPULATOR system. We will discuss about that. 

I think that the `open_manipulator_moving_state` is needed to make known to the user that some danger may be occur due to robot's moving. So, the `open_manipulator_moving_state` should be shown as stopped when the robot is being moved by a person.
Please share your opinions about this to us. And we need your contribution about this update.

gpwhd07, 2019-08-16:
We discussed this issue. It was judged more appropriate to maintain the current ecology.
The function of displaying this moving is to inform the user of danger.
And robotis_manipulator, which is the basis of open_manipulator_libs, should be able to perform all functions even when using motors other than Dynamixels. For this reason, it is difficult to modify the program to display data(122) from a motor (dynamixel).

This issue will be closed since there were no actions for a while. You can reopen this issue to show this issue to the users whenever. Thanks.

closed 2019-08-16 by gpwhd07