ROBOTIS-GIT/open_manipulator #105: Open Manipulator with 6 Joints

HaShaq, 2019-05-23:
Hi, 
iam working on a related issue. I want to extend the OM with two extra Dynamixel XM430. I was successful in controlling the 6dof OM with the open_manipulator_teleop. I can control single joints perfectly. Its just the IK that doesn't work properly. I think i changed everything according to the instructions mentioned in this issue: [(https://github.com/ROBOTIS-GIT/open_manipulator/issues/84)]. The arm moves, but not as expected.
What exactly do i need to change? Do i have to write my own kinematics or is the `kinematics.cpp` adapting somehow to the extra joints? 

Thank you :)

gpwhd07, 2019-05-27:
Hi @HaShaq 
Thanks for using our OpenMANIPULATOR.
We already made OpenMANIPULATOR SARA that is manipulator having 6DOF. The source code for controlling 6DOF OM is available at [open_manipulator_friends](https://github.com/ROBOTIS-GIT/open_manipulator_friends).

HaShaq, 2019-05-29:
Thank you @gpwhd07 ! This is exactly what i was looking for. I also noticed some false movements inside the Gazebo Simulation, especially with the gripper. It is not moving correctly. A bug maybe? 

Also, do you have any experience with the XM430-W210-T Servos? Iam using them for my OM instead of the recommended XM430-350-T. Does it make a difference? i noticed that the controller launch gives me this:
```
[ERROR] [DynamixelWorkbench] Failed to set time based profile!
[ERROR] Please check your Dynamixel firmware version (v38~)
```

MONICA-PALLA, 2019-06-03:
Hello @gpwhd07,
Could you please let me know the payload for 6DOF arm (OpenMANIPULATOR SARA) that you have made?

Thank you,
Monica, P.

gpwhd07, 2019-06-03:
@HaShaq 
Sorry for the late reply.
The XM430-W210-T also works fine.

> [ERROR] [DynamixelWorkbench] Failed to set time based profile!
> [ERROR] Please check your Dynamixel firmware version (v38 ~)

This error occurs because the firmware of Dynamixel is low.
Download the Wizard 2.0 and refer to the link below to update the Dynamixel firmware.
[e-manual Wizard 2.0](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/)

gpwhd07, 2019-06-03:
Hi @MONICA-PALLA 
OpenMANIPULATOR SARA is not a release version, it is just a test model.
So we did not measure the payload of OpenManIPULATOR SARA accurately,
but we estimate that it will be roughly 100g or less.

HaShaq, 2019-06-09:
Thank you very much for your help! Issue is solved :+1:

closed 2019-06-09 by HaShaq

mayank0621, 2023-08-20:
Hi I'm trying to use inverse Kinematics on Open manipulator-x (4 DOF)  using moveit2 ROS2 humble. I've successfully tested inverse kinematics on simulation by adding two virtual joint in urdf. But in actual hardware I'm facing issue. Any help would be appriaceted. Thanks in advance!!!!