ROBOTIS-GIT/open_manipulator #107: I bought OpenManipulator.

zebracheng, 2019-06-24:
I bought an OpenManipulator-X. How can I go about to change the source code on 
Arduino to make the Gripper Open wider.
I want to move the gripper to  90 instead of 0.009

zebracheng, 2019-06-24:
I'm using the Processing 3 software ( hand guide) to move the gripper open and close.
the Gripper is able to open very small angle. How can i go about to modify the source code to allow the
Gripper open wider angle. I need help or advice.

gpwhd07, 2019-06-25:
Hi @zebracheng 
Thank you for using our OpenMANIPULATOR-X.

The Unit of the gripper movement is 'm', not 'mm', and it means the moving distance of one gripper finger.
And the range of the grippper position is -0.009m to 0.009m. The gripper position of the open state is 0.009m, and it of close state is -0.009mm. 
So that the 0.009 means that the gripper fingers gap is about 4cm.
And that is maximum value. 
Please check that the gap is 4cm at the open state. 
If it is not 4cm, It may be a situation that the gripper dynamixel does not move more due to the current limit. 

If so, it is possible that the 4 Flange bush screws on the gripper are overtightened. Please loosen that a little.
Another way is to raise the current limit. please add below codes to source code.
(under the `open_manipulator.initOpenManipulator(platform_state);` of the `void setup()` function in the .ino file) 

```
STRING gripper_dxl_opt_arg[2];
void *p_gripper_dxl_opt_arg = &gripper_dxl_opt_arg;
gripper_dxl_opt_arg[0] = "Goal_Current";
gripper_dxl_opt_arg[1] = "300";
open_manipulator_.setToolActuatorMode(TOOL_DYNAMIXEL,p_gripper_dxl_opt_arg);

```
The value '300' can be changed in the range of 200 to 1193. Increasing this value increases the gripping force output by the Dynamixel.

zebracheng, 2019-06-25:
Thanks so much for getting me so quickly.
what I mean is that I will not use the gripper in my application. I will remove the gripper off the the

zebracheng, 2019-06-25:
i will remove the gripper off from the end of tool and leave the servo ID 15 attached on the end of tool. my application is to able  to move the server 15 to 0 degree or negative or pasitive 90 degree.  i did some testing under U2D2 hardware using the application i was able to move the ID15 to zero degree or +90 or -90 degree. My application is to pick up some small label and turn the label in landscape direction and apply label on the box. if this idea work I will by 3 to 4 Robots. I will try the code that you mentioned to see if it  will work. it is hard to send mail. is there any way I can chat on the phone with you. This is just and idea to see if the OpenMani Robot will work. Then later I will and modify the end of tooling and add IO on the Open CR using Arduino hardware. The Robot hardware is good but the software is limited me to move on. i have to prove that my concept is working then my manager will allow me to buy the pro version Robots because it has longer pay load and longer reach

zebracheng, 2019-06-25:
I remove the left and right grippers and use the hand guide software it is still move only 0.009 or -0.009.
so it not related to over tie or current. Some how the code is set to 0.009 and -0.009.  is there a way that you can show me how to change to larger value allow me to move to like 0 degree or +90 or -90 by using hand guide teach.

gpwhd07, 2019-06-25:
Hi again, 
Unfortunately, I can't directly contact you on phone or chat due to my another busy works, but I will reply asap on mails.

First of all, to apply your application, you have to add another tool such as vacuum tool at the end of dynamixel ID15. I think the vacuum tool is best choise to pick up the labels. Is that passible?
If you can, OpenMANIPUALTOR can work enough on your application.

Please refer [openMANIPULATOR-X emanual](http://emanual.robotis.com/docs/en/platform/openmanipulator_x/tool_modification/#vacuum-gripper) and [instrucable page](https://www.instructables.com/id/Vacuum-Gripper-System-Using-OpenCR/) about how to add and use vaccum tool to openManipulator-X.
And then the OpenMANIPULATOR source code is working based the ROBOTIS manipulator Library. please refer [e-manual](http://emanual.robotis.com/docs/en/software/robotis_manipulator_libs/) about this ROBOTIS manipulator Library.

zebracheng, 2019-06-25:
Hi Kim

(under the open_manipulator.initOpenManipulator(platform_state); of the void setup() function in the .ino file).

To add code where can I find the source file.

[External Email]

- CONFIDENTIAL-

This email and any files transmitted with it are confidential, and may also be legally privileged. If you are not the intended recipient, you may not review, use, copy, or distribute this message. If you receive this email in error, please notify the sender immediately by reply email and then delete this email.

zebracheng, 2019-06-25:
Hi Kim,
I found where to add the code .

open_manipulator_.setToolActuatorMode(TOOL_DYNAMIXEL,p_gripper_dxl_opt_arg);
is not define in the scope.
Help

gpwhd07, 2019-06-25:
sorry, it wasn't correct.
it was not `open_manipulator_.setToolActuatorMode(TOOL_DYNAMIXEL,p_gripper_dxl_opt_arg);`,
it is `open_manipulator.setToolActuatorMode(TOOL_DYNAMIXEL,p_gripper_dxl_opt_arg);`.

zebracheng, 2019-06-27:
Hi Kim,

The code that u provided that not help.
The R+ Manager is able to turn the ID15 0 degree to +90 or -90.
The suction cup code provided by Robotis is not working  because it did  not use ID15 servo .
 This make the arm short reach.
For my application I need  ID11 to ID15 on my project. Application needs  longer reach.
Is there any way that you can help to show me how to modify gripper ( Gripper open or Gripper close )code to 0 to + 90 or -90 instead 0.009 or -0.009.

[External Email]

gpwhd07, 2019-06-27:
You can not modify the information about the structure of the manipulator in the source code of open_manipulator_chain of the existing example. If you refer to open_manipulator_vacuum.h in open_manipulator_chain_vacuum, you will know how to modify the information.

The source code you are using in OpenCR is in the github repository at the link below.
https://github.com/ROBOTIS-GIT/OpenCR

Look at the link below.
https://github.com/ROBOTIS-GIT/OpenCR/blob/master/arduino/opencr_arduino/opencr/libraries/OpenManipulator/src/open_manipulator_libs/src/open_manipulator.cpp

The following code is the part that declares the structure of the manipulator.

```
  addTool ("gripper", // my name
          "joint4", // parent name
          math :: vector3 (0.126, 0.0, 0.0), // relative position
          math :: convertRPYToRotationMatrix (0.0, 0.0, 0.0), // relative orientation
          15, // actuator id
          0.010, // max gripper limit (0.01 m)
          -0.010, // min gripper limit (-0.01 m)
          -0.015); // Convert unit from `meter` to` radian`
```

`-0.015); // Convert unit from meter to radian `is the part that converts Dynamixel angle and input value of send function.
If you change this to 1.0, the Dynamixel will move as the input value.
Of course, you must also change the limit. This value is in radians.

The details of this, as I explained before, are listed below.
http://emanual.robotis.com/docs/en/software/robotis_manipulator_libs/#usage-tutorials

gpwhd07, 2019-08-09:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-08-09 by gpwhd07