ROBOTIS-GIT/open_manipulator #108: How to set ros controller to control a real arm?

Jiang765, 2019-07-18:
Hi! I am new to ROS and Open Manipulator. I just learned a bit from Moveit tutorials and want to control a real arm using Moveit. However, I found that making connection to a real arm needs to use ros controller. This information should be written to the `ros_controller.yaml` file under `open_manipulator_moveit/config` folder. I edited the arm setting using Moveit setup assistant and it automatically created a ros controller, shown as below. But later I checked the ros topic and service, there is actually no topic or service named follow_joint_trajectory or anything similar. I also tried to launch the `open_manipulator_controller.launch` with `use_moveit` option being true, the errors showed that there was no action coming up. 

So, I would like to know how to fix this problem. Or, may I ask if there are other easier ways that can be used to control the real arm by coding?

The related Moveit tutorial is: [](http://docs.ros.org/kinetic/api/moveit_tutorials/html/doc/controller_configuration/controller_configuration_tutorial.html)

Thank you very much!

[image]
[image]
[image]

gpwhd07, 2019-07-22:
Hi, @Jiang765 

OpenManipulator supports moveit, but it can run without moveit.
We recommend using e-manual to run without moveit. 
http://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_operation/#ros-operation

Nevertheless, if you want to use moveit to control, we recommend to use following command.
`$ roslaunch open_manipulator_controller open_manipulator_controller.launch use_moveit: = true`
Since moveit can not directly give the command value to the motor (dynamixel), we have to create a node for it. 
The open_manipulator_controller node can also support this function.
About how can you use moveit, please refer the [e-manual](http://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_operation/#moveit) and other closed issues.

Jiang765, 2019-07-23:
I see. Thank you so much!

closed 2019-08-09 by gpwhd07