ROBOTIS-GIT/open_manipulator #109: Open Manipulator with XL-320 servos

Tibeo, 2019-08-08:
Hello:),

Thanks for your time. I am using 5 XL-320 servos on Open Manipulator. I update the ID of 5 servos to  11~15 and the baud rate to 1000000 using Dynamixel Wizard 2.0. And the servos could be driven by using Dynamixel Workbench. 
However, when it comes to Open Manipulator, I could not make it. It seems that XL-320 should use  "_dynamixel::JointDynamixel_" instead of "_dynamixel::JointDynamixelProfileControl_" and I set the modes to "_position_mode_" but it comes out some errors. Any help is much appreciated.

Could you advice how to modify the codes to adapt XL-320 servos? I would like to use XL-320 servos therefore the manipulator could be small and light.

` //change to dynamixel::JointDynamixel due to XL-320 in _**"open_manipulator.cpp"**_ 

    actuator_ = new dynamixel::JointDynamixel();
    //actuator_ = new dynamixel::JointDynamixelProfileControl(control_loop_time);
    
    // Set communication arguments
    STRING dxl_comm_arg[2] = {usb_port, baud_rate};
    void *p_dxl_comm_arg = &dxl_comm_arg;
    
    // Set joint actuator id
    std::vector<uint8_t> jointDxlId;
    jointDxlId.push_back(11);
    jointDxlId.push_back(12);
    jointDxlId.push_back(13);
    jointDxlId.push_back(14);

    addJointActuator(JOINT_DYNAMIXEL, actuator_, jointDxlId, p_dxl_comm_arg);

    // Set joint actuator control mode
    STRING joint_dxl_mode_arg = "position_mode";

    void *p_joint_dxl_mode_arg = &joint_dxl_mode_arg;
    setJointActuatorMode(JOINT_DYNAMIXEL, jointDxlId, p_joint_dxl_mode_arg);

    /*****************************************************************************
    ** Initialize Tool Actuator
    *****************************************************************************/
    tool_ = new dynamixel::GripperDynamixel();

    uint8_t gripperDxlId = 15;
    addToolActuator(TOOL_DYNAMIXEL, tool_, gripperDxlId, p_dxl_comm_arg);

    STRING gripper_dxl_mode_arg = "position_mode";
    void *p_gripper_dxl_mode_arg = &gripper_dxl_mode_arg;
    setToolActuatorMode(TOOL_DYNAMIXEL, p_gripper_dxl_mode_arg);`

 **"roslaunch open_manipulator_controller.launch"**
**_Errors come as follows._**

SUMMARY
========

PARAMETERS
 * /open_manipulator/control_period: 0.01
 * /open_manipulator/moveit_sample_duration: 0.05
 * /open_manipulator/planning_group_name: arm
 * /open_manipulator/using_moveit: False
 * /open_manipulator/using_platform: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    open_manipulator (open_manipulator_controller/open_manipulator_controller)

auto-starting new master
process[master]: started with pid [4021]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a8b04352-b9e4-11e9-82a7-049226cd8812
process[rosout-1]: started with pid [4049]
started core service [/rosout]
process[open_manipulator-2]: started with pid [4060]
Joint Dynamixel ID : 11, Model Name : XL-320
Joint Dynamixel ID : 12, Model Name : XL-320
Joint Dynamixel ID : 13, Model Name : XL-320
Joint Dynamixel ID : 14, Model Name : XL-320
Gripper Dynamixel ID : 15, Model Name :XL-320
[ERROR] [TxRxResult] There is no status packet!
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
[INFO] Succeeded to init /open_manipulator
[ERROR] [TxRxResult] There is no status packet!
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
[ERROR] [TxRxResult] There is no status packet!
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
[ERROR] groupSyncRead getdata failed
...

rjshim, 2019-08-09:
@Tibeo 

Unfortunately, our package does not support XL320 model. Please use XM430 as written on our eManual (XL430 model can be also used but not recommended since we found motor overload issues when testing it).

Regards,
Ryan

Tibeo, 2019-08-09:
@rjshim Thanks for your information. It is a pity that I could not use XL-320 models. I have several XH430-W210 models and they are OK with the package. What I am trying to do is making a tiny manipulator. So I could refer to Dynamixel workbench and create a new package to motivate XL-320, right? Do you have any tips or suggestions on this? Thanks a lot.

rjshim, 2019-08-09:
@Tibeo 

Would you test it again with giving 7.4V instead of 12V that you might have given to motors? 
Check out the [link](http://emanual.robotis.com/docs/en/dxl/x/xl320/#specifications) to know the specification of XL-320.

Regards,
Ryan

Tibeo, 2019-08-11:
Hello Ryan @rjshim,

Do you mean I test XL-320 again using open manipulator with 7.4V? I did the test with 7.4V voltage. Anyway, I will go through the specification and test it again. 

You told me that open manipultor package could not support XL320. I am now a bit confused about whether it could support XL320. Would you help verfy it if you have XL-320 servos on hand? Thanks a lot.

Thanks and Regards,
Tibeo

rjshim, 2019-08-11:
@Tibeo 

We do not support each person's own customized robot so I recommended you use XM430 as instructed on our eManual. 

I thought the error code was from DXL-WB and one of the possible reasons you got that error code is because you might've given wrong voltage to motors. 

I only tested if DXL-WB works fine with XL-320 and 7.4V and it was fine. You said you tested om package with XL320 and 7.4V and if you are still having the same problem, unfortunately, it is beyond the scope of our support.

Regards,
Ryan

gpwhd07, 2019-09-09:
This issue has been closed as there weren't recent activities. Please feel free to reopen this thread if there's any opinion to throw. Thanks.

closed 2019-09-09 by gpwhd07