ROBOTIS-GIT/open_manipulator #111: Use U2D2 to controll OM failed

zhl017, 2019-08-12:
As title
My environment is PC(Ubuntu16.04 + kinetic) + U2D2 + SMPS2DXL + OM
What I already do :
1. set servos ID 11 to 15 and baudrate set 1000000mbps.
2. USB Latency Timer Setting finish

When I run `roslaunch open_manipulator_controller open_manipulator_controller.launch` 
Show this error 
[image]

But when I use old USBtoDynamixel, it can success work

gpwhd07, 2019-08-13:
hi @zhl017 
Thanks for using OM.
We need more information about your situation.
I think that problem may be due to U2D2.

Please check again that dynamixel successfully work in your hardware environments, by using the dynamixel wizard 2.0. 
http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_wizard2/

zhl017, 2019-08-13:
yes, it can work at wizard 2.0.
I use two U2D2, both happened this problem.
by the way, I use XM430-W350-R instead XM430-W350-T

rjshim, 2019-08-13:
@zhl017 

Would you test if [this](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_workbench/#controllers) work ? 

Regards,
Ryan

zhl017, 2019-08-13:
good to work ...
[image]

rjshim, 2019-08-13:
@zhl017 

Please test the next one (5.1.2) to see if the motor controller works properly.
You need to set your motor ids to 1,2 (you need two motors for this example)

Regard,
Ryan

zhl017, 2019-08-13:
Can't work and I didn't see `/dynamixel_command`..
I clone wrong package ??
[image]
[image]

rjshim, 2019-08-13:
@zhl017 

Please do the followings
1. Update your motor firmware with DXL-Wizard
2. Download the latest version of WB package

and let me know if it still doesn't work.

Regards,
Ryan

zhl017, 2019-08-13:
[image]

zhl017, 2019-08-13:
I use `git clone https://github.com/ROBOTIS-GIT/dynamixel-workbench.git` yesterday
I think it's the latest version.
and my motor firmware also the latest
[image]

rjshim, 2019-08-13:
@zhl017 

Would you test if [this tutorial](http://emanual.robotis.com/docs/en/software/dynamixel/dynamixel_sdk/sample_code/cpp_sync_read_write_protocol_2_0/#cpp-sync-read-write-protocol-20) works?

Regards,
Ryan

zhl017, 2019-08-14:
sorry for reply late.
I try sync_read_write C++ example
Motor can move one times and then show failed.
[image]

rjshim, 2019-08-14:
@zhl017 

1. Would you connect your U2D2 to a USB hub and connect to it to your computer, and test it again? 
2. Check if you really changed USB latency](http://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_setup/#usb-latency-timer-setting) and test it again.
3. If it still doesn't work, the problem might be on your U2D2, so just buy a new one.

Regards,
Ryan

zhl017, 2019-08-15:
So strange, I change another PC, it can work now !!

thanks.

closed 2019-08-15 by zhl017