ROBOTIS-GIT/open_manipulator #112: service question: open_manipulator/goal_task_space_path_from_present_position_only

junghyeonchiul, 2019-08-12:
Hello! 
I'm interested in reinforcement learning and i think that open_manipulator is a good platform to merge with robot, ROS and simulation(GAZEBO).

So i import open_manipulator on GAZEBO and search any topic or service that help me to operate the manipulator arbitrarily. 
Consequently, i found the service named "open_manipulator/goal_task_space_from_present_position_only". It seems like excellent ros service through which i instruct robot to move a discrete distance relative to robot's present position.
I see the message file used in that service communication, it takes a  planning_group_name, end_effector_name, kinematics_pose, path_time then outputs the is_planned that i think it means the making trajectory is complete. 

But when i inputs the position that robot can't reach, the output(is_planning) is still 'True"!!
I search the section that process the service communication and i found open_manipulator_controller/src/open_manipulator_controller.cpp file and noticed line 330~343.
In the function named goalTaskSpaePathPositionOnlyCallback, it always assign is_planned with True no matter what the position is reached or not reached by robot.
here's a sample of that file
~~~
 bool OpenManipulatorController::goalTaskSpacePathPositionOnlyCallback(open_manipulator_msgs::SetKinematicsPose::Request  &req,
                                                                      open_manipulator_msgs::SetKinematicsPose::Response &res)
{
  Eigen::Vector3d position;
  position[0] = req.kinematics_pose.pose.position.x;
  position[1] = req.kinematics_pose.pose.position.y;
  position[2] = req.kinematics_pose.pose.position.z;

  open_manipulator_.makeTaskTrajectory(req.end_effector_name, position, req.path_time);

  res.is_planned = true;
  return true;
}
~~~

Why is planned is assigned with True??? I'm so confused.
I just want to get output meaning whether this position is reachable or not reachable, when i command the robot to move to some position through ros service.
please help me. Thank you!

gpwhd07, 2019-08-13:
hi @junghyeonchiul 

Thanks for using OpenManipulator-X.
You are right. The service always returns "is_plannd" as true regardless of whether the plan succeeds because the return type of the makeTaskTrajectory function is void. 

In the case of a jointspace planning, we would soon fix the problem that "is_plannd" is always returned true.

When plannig a path in the taskspace, only the path of the location of the taskspace(world coordinates) is planned in advance, IK is solved every period while moving.
For this reason, we can not know whether the position can be reached by robot, before the robot move.

You can use makeJointTrajectory service for reinforcement learning. Many research of reinforcement learning use the motion in the joint space for reducing the effect by the failure, the performance or the accuracy of the Ik solving.

If you still want to use taskspace planning, you need to modify the source code, "open_manipulator_controller.cpp" and open_manipulator.cpp" in open_manipulator_libs.

gpwhd07, 2019-08-13:
hi again @junghyeonchiul 

Please check the `develop` branch of `open_manipulator` and `robotis_manipulator`.
I just update, so that `goalJointSpacePathToKinematicsPoseCallback` will be successfully work.
But `goalTaskSpacePathPositionOnlyCallback` return is_planned to true  whether IK solving success or fail, due to the reason I said above.

If the target joint value for every moving period is solved in advance and stored in a buffer and then sent to the Dynamixel, success or failure can be returned accurately.

junghyeonchiul, 2019-08-13:
hi @gpwhd07 

Thank for your kind answer.
I see the branch and check your modification about ~~~open_manipulator_controller~~~'s callback functions where ~~~is planned~~~ is assigned with True/False in ~~~if~~~ sentence,

Additionally, I have a question about how to use this updated code. Just download the modified file and move them to ~~~catkin_ws/src~~~ then ~~~catkin_make~~~???

Lastly, In updated ~~~open_manipulator_controller~~~'s callback functions, ~~~goalTaskSpacePathCallback~~~ functions also assign ~~~is_planned~~~ with True/False according to ~~~if~~~ sentence. I think this function seem to work well now. But your above answer look like that you only modify ~~~goalJointSpacePathCallback~~~s. I wonder whether only ~~~goalJointSpacePathCallback~~~s work well or all of trajectory(joint/Task) callback functions work well.

Thank you.

junghyeonchiul, 2019-08-13:
hello again @gpwhd07 .

I downloads your updated version of open_manipulator_controller & robotis_manipulator and catkin_make them in my catkin_ws/src. 
And i open rqt-service-caller and use them. As you refer in above your answer, `goalJointSpacePathToKinematicsPoseCallback` work well (it outputs False when i inputs impossible pose for robot). But other callback functions seem like that they doesn't work well yet. They still outputs True regardless of position i inputs.

Do i understand your modifications rightly?

gpwhd07, 2019-08-13:
Hi @junghyeonchiul 

As mentioned earlier, I have modified only `goalJointSpacePathToKinematicsPoseCallback` works correctly, but not `goalTaskSpacePathCallback`.
`goalTaskSpacePathCallback` is return `is_planned` as `false`, when only end_effector_name was not correct. 

Whether planning success or fail can be known in `goalTaskSpacePathCallback` function, because IK is not solved in this function and is solved in `process` function in every loop time.
On the other hand, in the case of `goalJointSpacePathToKinematicsPoseCallback`, IK is solved in this function.

junghyeonchiul, 2019-08-13:
Hi @gpwhd07 

Ok. i got it. That's a reason why `goalTaskSpacePathCallback` function don't work.
So now, i have to use `goalJointSpacePathToKinematicsPoseCallback` function for operating robot.

Thanks for your detail explanation.

gpwhd07, 2019-08-13:
I'm glad it helped you.
In practice, joint tarajectory may be more effective for reinforcement learning. This is because errors by the Ik solving to learning can adversely affect learning.
I hope your research will do well. If you have any good results, please let us know and we will be happy.

gpwhd07, 2019-08-16:
This issue will be closed. If you have any question or suggest, please ask to me any time. Thanks :)

closed 2019-08-16 by gpwhd07