ROBOTIS-GIT/open_manipulator #113: error about industrial_trajectory_filters

hpf9017, 2019-08-21:
when I use:
roslaunch open_manipulator_controller open_manipulator_controller.launch use_moveit:=true

I get such error about the industrial_trajectory_filters:

[ INFO] [1566376953.058954462]: Initializing OMPL interface using ROS parameters
[ INFO] [1566376953.082171635]: Using planning interface 'OMPL'
[ERROR] [1566376953.084384593]: Exception while loading planning adapter plugin 'industrial_trajectory_filters/UniformSampleFilter': Failed to load library /opt/ros/kinetic/lib/libindustrial_trajectory_filters.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libmoveit_robot_trajectory.so.0.9.17: cannot open shared object file: No such file or directory)
[ERROR] [1566376953.084662325]: Exception while loading planning adapter plugin 'industrial_trajectory_filters/AddSmoothingFilter': Failed to load library /opt/ros/kinetic/lib/libindustrial_trajectory_filters.so. Make sure that you are calling the PLUGINLIB_EXPORT_CLASS macro in the library code, and that names are consistent between this macro and your XML. Error string: Could not load library (Poco exception = libmoveit_robot_trajectory.so.0.9.17: cannot open shared object file: No such file or directory)
[ INFO] [1566376953.088487670]: Param 'default_workspace_bounds' was not set. Using default value: 10
[ INFO] [1566376953.089441370]: Param 'start_state_max_bounds_error' was set to 0.1
[ INFO] [1566376953.090782223]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1566376953.091430091]: Param 'start_state_max_dt' was not set. Using default value: 0.5
[ INFO] [1566376953.092048799]: Param 'jiggle_fraction' was set to 0.05
[ INFO] [1566376953.092904101]: Param 'max_sampling_attempts' was not set. Using default value: 100
[ INFO] [1566376953.093039423]: Using planning request adapter 'Add Time Parameterization'
[ INFO] [1566376953.093084946]: Using planning request adapter 'Fix Workspace Bounds'
[ INFO] [1566376953.093098643]: Using planning request adapter 'Fix Start State Bounds'
[ INFO] [1566376953.093114018]: Using planning request adapter 'Fix Start State In Collision'
[ INFO] [1566376953.093128302]: Using planning request adapter 'Fix Start State Path Constraints'

gpwhd07, 2019-09-09:
Hi @hpf9017 

I can guess that error occur because industrial core was not run.
Did you install the ros-kinetic-industrial-core package?

`sudo apt-get install ros-kinetic-industrial-core`
or 
`cd catkin_ws/src && git clone https://github.com/ros-industrial/industrial_core.git`

please check this.

hpf9017, 2019-09-11:
thanks!
first, i install by apt-get, but it doesn't work
then I download the source code and build.
it solves my problem

gpwhd07, 2019-09-11:
That's good.
Thanks for your feedback.
This issue will be closed. If you have the other question or suggest, please ask to me any time. Thanks :)

closed 2019-09-11 by gpwhd07