ROBOTIS-GIT/open_manipulator #115: A question about open_manipulator_x in ROS gazebo simulation

Sunchanghao, 2019-08-26:
I meet a big problem when I use open_manipulator_X during the process of gazebo simulation.
I put the problem's video in annex.
because I install the manipulator on the PX4 drone,the control period of the PX4 is 250HZ,so I set the parameter in empty.world like this:
    
 <real_time_update_rate>250.0</real_time_update_rate>
      <max_step_size>0.004</max_step_size>
      <real_time_factor>1</real_time_factor>

I also check the .xacro files and I only find one parameter in open_manipulator.gazebo.xacro like that:

  
  <gazebo>
    <plugin name="gazebo_ros_control" filename="libgazebo_ros_control.so">
      <robotNamespace>open_manipulator</robotNamespace>
      <controlPeriod>0.004</controlPeriod>
      <robotSimType>gazebo_ros_control/DefaultRobotHWSim</robotSimType>
      <legacyModeNS>true</legacyModeNS>
    </plugin>
  </gazebo>

 I modified these two files and then,the question arises:manipulator shaking in the gazebo.
and I find that the lower update_rate,the bigger shake.
I try to modify PX4 files,but only gazebo simulation rate is 250HZ can start the gazbeo.

so my question is whether users can modify the control period of the open_manipulator_x,if we can,than how to do ? ^_^
if we can not modify the parameter 1000hz,when I use material object,will I have the same problem?
This little problem has not been solved for two weeks.I would great aooerciate you if you could help me.
thank you!

gpwhd07, 2019-09-09:
Hi @Sunchanghao 
I understand the problem.
That vibration problem occurs in the gazebo ros controller.
The reduced control rate prevents the controller from converging properly. You can try to reset the PID gains in gazebo controller to solve this problem. These PID gains are set in open_manipulator_gazebo/config/open_open_manipulator_controller.yaml in open_manipulator_simulation package. 
Plaese try it.

Sunchanghao, 2019-09-09:
@gpwhd07 Thank you! I have realized that I need reset the PID gains in  .yaml 
like you say, after I try to reset the PID gains,the model do not shake again, even more stable than the original model. 
Thank you!

gpwhd07, 2019-09-09:
@Sunchanghao 
That's good. 
We look forward to seeing your project perform well.

closed 2019-09-09 by gpwhd07