ROBOTIS-GIT/open_manipulator #117: Error when using processing GUI

rx78kok, 2019-09-04:
Hi, i have problems when using the processing GUI to move open manipulator based on the manual that was given. I am using the processing code for open manipulator chain given in this site. This is the error im getting. This error happen when i tried to connect the processing with the port that should be connected with my openCR board.

controlP5.ControlBroadcaster printMethodError
SEVERE: An error occured while forwarding a Controller event, please check your code at Controller_OnOff
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at sun.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at controlP5.ControlBroadcaster.invokeMethod(Unknown Source)
	at controlP5.ControlBroadcaster.callTarget(Unknown Source)
	at controlP5.ControlBroadcaster.broadcast(Unknown Source)
	at controlP5.Controller.broadcast(Unknown Source)
	at controlP5.Toggle.setState(Unknown Source)
	at controlP5.Toggle.mousePressed(Unknown Source)
	at controlP5.Controller.setMousePressed(Unknown Source)
	at controlP5.ControllerGroup.setMousePressed(Unknown Source)
	at controlP5.ControlWindow.mousePressedEvent(Unknown Source)
	at controlP5.ControlWindow.mouseEvent(Unknown Source)
	at sun.reflect.GeneratedMethodAccessor6.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at processing.core.PApplet$RegisteredMethods.handle(PApplet.java:1432)
	at processing.core.PApplet.handleMethods(PApplet.java:1634)
	at processing.core.PApplet.handleMouseEvent(PApplet.java:2742)
	at processing.core.PApplet.dequeueEvents(PApplet.java:2645)
	at processing.core.PApplet.handleDraw(PApplet.java:2486)
	at processing.awt.PSurfaceAWT$12.callDraw(PSurfaceAWT.java:1547)
	at processing.core.PSurfaceNone$AnimationThread.run(PSurfaceNone.java:313)
Caused by: java.lang.NullPointerException
	at Chain$ChildApplet.Controller_OnOff(Chain.java:894)
	... 24 more

gpwhd07, 2019-09-09:
Hi @rx78kok 
I guess that the error occur due to wrong installation of the controlP5 plugin. 
Please reinstall the controlP5 plugin as accoding to the [e-manual](http://emanual.robotis.com/docs/en/platform/openmanipulator_x/opencr_operation/#setup-processinggui)
If it still happen, it can be GPU driver problem, please test it in the other PC.

rx78kok, 2019-09-09:
I see. I will try reinstall the plugin. If it still the same happen then i will try update gpu driver. Will be get back here again as soon as i done test it. Thank you.

rx78kok, 2019-09-11:
Hi. Is it possible for the same reason causing the controller freeze or stuck when both control interface windows appears?

gpwhd07, 2019-09-11:
Hi @rx78kok 
Yes I think it is possible.
Both graphic interface are implemented by control p5 plugin in our processing IDE program, so that the problem causing control interface windows freeze is most likely to occur with the control p5 plugin.

And there have been many reports that the first time running processing IDE source code, it takes a lot of time to run the graphical interface. Therefore, if no error occurs, wait until the graphical interface is fully up.

closed 2019-09-19 by rx78kok

Indruino-hieu, 2020-03-13:
Hi,all! 
i'm try everything but its not working. 
My error : java.lang.OutofMemoryError:GC overhead limit exceeded!
could you help me! please!