ROBOTIS-GIT/open_manipulator #124: Screws Problem

zarpli, 2019-10-02:
We follow the instructions in the following manual:

http://www.robotis.com/service/download.php?no=1255

On page 7, step 13. Indicates that screws WB-M2.5x12 must be used to
assemble FR12-H104K to Gripper. But these screws are not long enough to do this.

Are we doing something wrong?, or is there an error in the manual?

Best regards

gpwhd07, 2019-10-04:
Hi @zarpli 

As you said, the WB-M2.5x12 is short enough to assemble the gripper.
I am aware of this issue and it has changed from WB-M2.5x12 to WB-M2.5x17.
The manual at that [link](http://www.robotis.com/service/download.php?no=1255) also changed to WB-M2.5x17 a few months ago.

Please directly contact about support for customers who have purchased prior to the change. 
As US Branch Office is responsible for Sales & Operations in the North, Central, and South American region. Please direct your issue and question to america@robotis.com and our branch office will help assist you.

zarpli, 2019-10-05:
Hi @gpwhd07,

I asked for help since June 10 of this year, but I had no response from ROBOTIS Technical Support / Customer Support.

I will insist again, thank you very much for the clarification and support!

gpwhd07, 2019-10-07:
Hi @zarpli 

Sorry for the inconvenience. Please contact the US branch again.

closed 2019-10-08 by gpwhd07