ROBOTIS-GIT/open_manipulator #125: the right input about the service "/open_manipulator/moveit/set_joint_position" ?

hpf9017, 2019-10-03:
I try many format of the right input about the service, but fail.

gpwhd07, 2019-10-04:
Hi @hpf9017 

Thanks to use our OpenMANIPULATOR-X.

If the plan fail even you set the position(x,y,z) robot can go, please check the 'max_accelerations_scaling_factor', 'max_velocity_scaling_factor' and 'tolerance'. the default value of the 'max_accelerations_scaling_factor' and 'max_velocity_scaling_factor' is 1.0. 
I recommend to set the 'tolerance' to 0.005~0.05. If you set that correctly, you can get response "True".

In addition, if you call only the 'moveit/set_kinematics_pose' service, the robot will not move even the plan succeed. After the plan succeed, you need to publish a empty topic message 'execute_trajectory/goal' have moveit_msgs/ExecuteTrajectoryActionGoal type.  Then robot will move to follow the planed trajectory.

Thanks.

hpf9017, 2019-10-04:
@gpwhd07 my question is about how to write the cmd value, for example,
rosservice call /open_manipulator/moveit/set_joint_position "planning_group: ''
joint_position:
  joint_name:
  - ''
  position:
  - 0
  max_accelerations_scaling_factor: 0.0
  max_velocity_scaling_factor: 0.0
path_time: 0.0"

what is the right format of the rosservice cmd, especially, the in joint_name and the position?
I'm sorry that I made it unclear.

gpwhd07, 2019-10-07:
Hi, @hpf9017 

First, please read again [e-manual](http://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_operation/#moveit).

- To use the `/open_manipulator/moveit/set_joint_position` service, you must first enable moveit when you run the open_manipulator_controller.launch file. Run with the `use_moveit` parameter set to `true`.

- We also provide a control method that does not use moveit. Please refer to the [e-manual](http://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_operation). You can set following. `/open_manipulator/goal_joint_space_path`

> planning_group : ''
> joint_position:
> joint_name : ['joint1','joint2','joint3','joint4']
> position : [0.0,0.0,0.0,0.0]
> max_acceleration_scaling_factor : 0.0
> max_velocity_scaling_factor : 0.0
> path_time 2.0

- Since open_manipulator-x has 4 degrees of freedom, not all orientations can be implemented. To use moveit, change the `position_only_ik` parameter in the `open_manipulator/open_manipulator_moveit/config/kinematics.yaml` file to `True`.

- In the `/open_manipulator/moveit/set_joint_position`, planning_group : `arm`

rjshim, 2020-01-09:
As there have been no updates, I will close this issue. 
Ryan

closed 2020-01-09 by rjshim