ROBOTIS-GIT/open_manipulator #126: Freezing during operation

Krrrl, 2019-10-11:
I have encountered an issue while working with my new OpenManipulator.
During operation of the physical platform from keyboard input, the controller stops taking inputs.
It seems that the open_manipulator_controller package freezes when given a rapid sequence of task-space inputs from teleop_keyboard.
I have followed the [emanual](http://emanual.robotis.com/docs/en/platform/openmanipulator_x/overview/) to the point, and also updated motor firmware through the DynamixelWizard2 application.

The application is launched as follows:
roscore
roslaunch open_manipulator_controller open_manipulator_controller.launch
roslaunch open_manipulator_teleop open_manipulator_teleop_keyboard.launch

The symptoms of the bug is:
-teleop_keyboard console freezes(most recent input is not handled)
-Manipulator freezes in place, with motor-torque enabled.
-The open_manipulator_controller topic: /open_manipulator/states/open_manipulator_moving_state = "IS_MOVING", and it never changes.
-The light on the U2D2 turns static green.
-The controller needs to be killed and launched again, at which point it produces error messages to the console(see below).

My initial thoughts is that this might be due to a hardware error between the U2D2 and open_manipulator_controller package?
Have you encountered something like this before?

I have asked another university affiliate to replicate the issue on his platform, but he has been unable to get the same results.

Here is the error output to console after relaunch of the controller package:

krrrl@gravel:~$ roslaunch open_manipulator_controller open_manipulator_controller.launch 
... logging to /home/krrrl/.ros/log/e6b87194-e50c-11e9-a545-989096ccd623/roslaunch-gravel-23524.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://localhost:37153/

SUMMARY
========

PARAMETERS
 * /open_manipulator/control_period: 0.01
 * /open_manipulator/moveit_sample_duration: 0.05
 * /open_manipulator/planning_group_name: arm
 * /open_manipulator/using_moveit: False
 * /open_manipulator/using_platform: True
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    open_manipulator (open_manipulator_controller/open_manipulator_controller)

ROS_MASTER_URI=http://localhost:11311

process[open_manipulator-1]: started with pid [23542]
Joint Dynamixel ID : 11, Model Name : XM430-W350
[ERROR] [DynamixelWorkbench] Failed to set time based profile!
[ERROR] Please check your Dynamixel firmware version (v38~)
[ERROR] [RxPacketError] Writing or Reading is not available to target address!
[ERROR] Please check your Dynamixel firmware version
Joint Dynamixel ID : 12, Model Name : XM430-W350
[ERROR] [DynamixelWorkbench] Failed to set time based profile!
[ERROR] Please check your Dynamixel firmware version (v38~)
[ERROR] [RxPacketError] Writing or Reading is not available to target address!
[ERROR] Please check your Dynamixel firmware version
Joint Dynamixel ID : 13, Model Name : XM430-W350
[ERROR] [DynamixelWorkbench] Failed to set time based profile!
[ERROR] Please check your Dynamixel firmware version (v38~)
[ERROR] [RxPacketError] Writing or Reading is not available to target address!
[ERROR] Please check your Dynamixel firmware version
Joint Dynamixel ID : 14, Model Name : XM430-W350
[ERROR] [DynamixelWorkbench] Failed to set time based profile!
[ERROR] Please check your Dynamixel firmware version (v38~)
[ERROR] [RxPacketError] Writing or Reading is not available to target address!
[ERROR] Please check your Dynamixel firmware version
Gripper Dynamixel ID : 15, Model Name :XM430-W350
[ERROR] [DynamixelWorkbench] Failed to set velocity based profile!
[ERROR] Please check your Dynamixel firmware version (v38~)
[ERROR] [RxPacketError] Writing or Reading is not available to target address!
[ERROR] Please check your Dynamixel firmware version
[INFO] Succeeded to init /open_manipulator

gpwhd07, 2019-10-11:
Hi @Krrrl 

If the green led of U2D2 is still on, I think it is a communication problem between U2D2 and PC.
Please check again the usb latency ([e-manual](http://emanual.robotis.com/docs/en/platform/openmanipulator_x/ros_setup/#usb-latency-timer-setting)).

Krrrl, 2019-10-11:
Hi @gpwhd07 

Thank you for your very fast reply!

I have replicated the error, and it is in fact all(red, green and blue) the leds that are on. when the controller is frozen.

I have checked the usb latency as you suggested, and running 
$ cat /sys/bus/usb-serial/devices/ttyUSB0/latency_timer
returns 1.

Is there anything else you could think of that might cause this problem?
Or, any other way to authenticate that it indeed is a latency issue?

gpwhd07, 2019-10-14:
Hi @Krrrl 

The Teleop console stops because it is waiting for a response from the controller node. And the reason why the controller node is stopped is most likely due to and no response from dynamixel. If it's not a USB problem, there is a good chance that the ttl cable between dynamixel and U2D2 is damaged. Please check the cable.

In addition, errors that occur when you run the controller again occur when you run the controller with the dynamixel's torque enabled (some items in the dynamixel cannot be written with the torque enabled). However, because these items are in the EEPROM Area, they will already be saved when you run the controller before. Therefore the errors are not a problem.

Krrrl, 2019-10-27:
Hi again @gpwhd07 

Sorry for my very late response.
I will replace the TTL cable with one of the extra ones from the kit, and attempt to replicate the error.
I will check this sometime this week, and let you know as soon as I have any results.

Thank you for your quick replies, I'll check in again in a couple of days!

closed 2020-01-09 by rjshim

ariccspstk, 2023-05-09:
@Krrrl Did you managed to solve your issue using a new TTL cable?

sasilva1998, 2023-07-28:
Hi, I am having the same issue, has anyone been able to solve it?

ChangAnCN, 2024-04-23:
Same issue, Any help?

gholibqasobov, 2024-05-18:
HI there, has anyone solved this issue? Am having the same trouble