ROBOTIS-GIT/open_manipulator #127: smooth operation question

hyunoklee, 2019-10-15:
Hello. I have a some question.
Please let me know if you know how to solve it.
I make code that rotates Joint1 of the open manipulator per 0.1 seconds like below code.

Operating is ok. But I want smooth motion.  
After finish one cycle Action ( 0.1 sec ) , The  motion is not smooth  at the start point of the next cycle Action .  
Is there any way to make motion of arm  smoothly  at  start of the next cycle?
I will apply the code to the tracking operation, so I don't want reduce the duration ( 0.1 sec ) for immediate response. 
If you know solution, please help me.  :) 

class trackingAction(smach.State):
    def __init__(self):
        ########skip code ########################################
	self.set_joint_position_from_present = rospy.ServiceProxy('/open_manipulator/goal_joint_space_path_from_present', SetJointPosition)

    def execute(self, userdata):
         while True :
            try: 
	        joint_position = JointPosition()
	        planning_group = userdata.input_planning_group
	        joint_position.joint_name = ['joint1','joint2','joint3','joint4']
                joint_position.position =  [0.3, 0, 0, 0]
                path_time = 0.1
                resp1 = self.set_joint_position_from_present(planning_group,joint_position, path_time)
	        rospy.sleep(0.1) 
            except rospy.ServiceException, e:
                print "Service call failed: %s"%e
        return 'succeeded'

gpwhd07, 2019-10-16:
Hi @hyunoklee 

When the controller node receives the `/open_manipulator/goal_joint_space_path_from_present` service, it makes a trajectory that stops exactly at that path_time. Therefore, the robot stops every 0.1 seconds (velocity = 0), and the whole trajectory is not a smooth.

Unfortunately,  it does not provide a solution(ROS service) to smoothly connects the two trajectories .

Set `path_time` higher (such as 0.2s) than now(0.1s) , and send the service every 0.1 seconds. The trekking speed will be slowed down, but if the target's position is updated every 0.1 seconds. The manipulator will track the target smoothly and continuously because the velocity is not zero.

hyunoklee, 2019-10-16:
I think that's a good solution. I will apply it later. Thank you ~

hyunoklee, 2019-10-16:
Close

closed 2019-10-16 by hyunoklee