ROBOTIS-GIT/open_manipulator #129: I have a question about goal_drawing_trajectory service.

hyunoklee, 2019-11-01:
hello ~ 
I want to draw a heart using the /open_manipulator/goal_drawing_trajectory. 
I think I can use open_manipulator_msgs/SetDrawingTrajectory Service. 
The service need below parameter.  

open_manipulator_msgs/SetDrawingTrajectory.srv
string end_effector_name
string drawing_trajectory_name
float64[] param
float64 path_time
---
bool is_planned

But I don't know how do I set value of the param to draw a heart.
E-manual just descript param  as " Parameters required for the drawing trajectory generation."

I would really appreciate if you give me some example about how do I set value of param of SetDrawingTrajectory.srv  to draw a heart.
And one more question, Does goal_drawing_trajectory API support only circle, the rhombus, the heart, and the straight line ? 
Is it possible for user to draw custom drawing by goal_drawing_trajectory  or other service ?

rjshim, 2019-11-21:
@hyunoklee 

Sorry for the late reply. I am not sure if you would still need help with this issue, but if you do, refer to the links below.

1. Drawing: [Link](https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_control_gui/src/qnode.cpp)
Check the function 'setDrawingTrajectory', then you will know how to use it. 

2. Custom trajectory: [Link](https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_libs/src/custom_trajectory.cpp)
It won't be easy to additionaly add one to the current architecture but it won't be that difficult to just change one to what you want to draw (e.g., modify initRhombus and drawRhombus functions and make functions for drawing a triangle. it's just simple math!)

Regards,
Ryan

minwoominwoominwoo7, 2019-11-21:
Thank you , I will try it.

hyunoklee, 2019-11-21:
Thank you. :) 
I will try it.~

closed 2019-11-21 by hyunoklee