ROBOTIS-GIT/open_manipulator #133: Is there a dynamic model program here

Sunchanghao, 2019-11-18:
Hello
I want to modify the size of open_manipulator_x 
so is there a dynamic model program here that I can modify the parameter?
THX

rjshim, 2019-11-19:
What do you mean dynamic model program? You want to change the size of robot links?

Ryan

Sunchanghao, 2019-11-19:
hello
The manipulator is mounted on the drone

so I need to calculate the center of gravity changes
thus I need to model the motion of the manipulator～

rjshim, 2019-11-19:
@Sunchanghao 

Please rerfer to [this](https://github.com/ROBOTIS-GIT/open_manipulator/blob/master/open_manipulator_description/urdf/open_manipulator.urdf.xacro)

Regards,
Ryan

closed 2019-12-06 by rjshim

Sunchanghao, 2019-12-11:
@rjshim Thanks
I have a question,is the calculation of the Denavit-Hartenberg(D-H) of the manipulator done in Movelt!?