ROBOTIS-GIT/open_manipulator #134: Chain Vaccum in Current Based Position Mode

morusa, 2019-12-03:
Hi,

I relied on the open_manipulator_chain_vaccum example and it works fine.

but I want to use the "current_based_position_mode" mode, to protect the joints because it will be teleoperated with a joystick, the users will be children and will not take care of the limits.

in "open_manipulator_vaccum.h" I changed the line:

STRING joint_dxl_mode_arg = "position_mode";

for this one:

STRING joint_dxl_mode_arg = "current_based_position_mode";

but the robot arm does not move, it only has a bit of blocking, but it can be moved with the hands.

I think I should set some current parameter, but where can I do it?

can you guide me?

gpwhd07, 2019-12-04:
Hi @morusa 

You can set the current limit as `Goal Current(102)` item of the dynamixel control table. This value sets a limit to current in `Current-based Position Control mode`.

On the our open_manipulator code, you can set this control table item (also the other any items) as using following codes.

```
std::vector<uint8_t> jointDxlId;
jointDxlId.push_back(1);
jointDxlId.push_back(2);
jointDxlId.push_back(3);
jointDxlId.push_back(4);
STRING joint_dxl_item_arg = "Goal_Current";
void *p_joint_dxl_item_arg = &joint_dxl_item_arg;
setJointActuatorMode(JOINT_DYNAMIXEL, jointDxlId, p_joint_dxl_item_arg);
```

morusa, 2019-12-09:
hi

The use of the robot in Current-based Position Control mode works well, I was able to limit the torque to the required parameters.

But there is a problem of "tremors." These tremors according to my analysis are due to the "backlash" of the servomotor gear.

Please watch the following video. 

https://youtu.be/DCceEq14ttU

please, with all the experience you have, what parameters could you adjust to eliminate this trembling behavior?

If I return the robot to position_mode mode the behavior is perfect, but how can I monitor the current that each servomotor is demanding? In order to protect the servomotors when the user tries to move the arm in a direction where an obstacle is found.

closed 2019-12-10 by morusa