ROBOTIS-GIT/open_manipulator #135: Problem with Current-based Position Control mode

morusa, 2019-12-10:
Hi

I am use the robot in <code>Current-based Position Control mode</code> works well, I was able to limit the torque to the required parameters.

But there is a problem of "tremors."  Please watch the following video.

https://youtu.be/DCceEq14ttU

These tremors according to my analysis are due to the "backlash" of the servomotor gear.

please, with all the experience you have, what parameters could you adjust to eliminate this trembling behavior?

If I return the robot to <code>position_mode</code> the behavior is perfect, but how can I monitor the current that each servomotor is demanding? In order to protect the servomotors when the user tries to move the arm in a direction where an obstacle is found.

rjshim, 2020-01-09:
@morusa 

You can change the position PID gains for your motors with Dynamixel Wizard. If the d gain is set to 0, increase it to reduce the tremors. Test it and let us know the result. 

Regards,
Ryan

rjshim, 2020-01-14:
As there have been no updates, I will close this issue.

Ryan

closed 2020-01-14 by rjshim